bambulab/BambuStudio #2970: BUG NELLA RIPRODUZIONE DELLA FILETTATURA

Stringa011, 2023-11-26:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Printer

bambu p1s

### How to reproduce

importando un file 3mf creato con fusion 360, non viene riportata la filettatura dell'oggetto. E di conseguenza quando si stampa non viene riprodotta la filettatura

### Actual results

quando il file viene caricato non riproduce la filettatura dell'oggetto rendendo impossibile eseguire la stampa

### Expected results

dopo aver caricato il file su bambu studio la filettatura dell'oggetto deve comparire sul modello per poi essere stampata

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-11-27:
翻译
导入使用 fusion 360 创建的 3mf 文件时，不会报告对象的线程。因此，在打印时，不会再现线程

加载文件时，它不会重现对象的线程，从而导致无法打印

将文件上传到 bambu studio 后，对象的线必须出现在模型上才能打印

bambu123, 2023-11-27:
@Stringa011 
Can help provide 3mf files and some interface screenshots to more clearly explain the problem?