bambulab/BambuStudio #2971: Pressure Advvanced for custom filament not available in BL Studio

alienm16, 2023-11-26:
### Bambu Studio Version

1.8.0.62

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11 23h2 build 22631.2715

### Additional system information

Alienware m16
i9-13900HX 13th generation
Nvidia RTC 4070

### Printer

Bambu Lab X1CC with firmware 1.06.06.03

### How to reproduce

Create a custom filament in BL Studio
Make a automatic dynamic flow calibration for the filament
Save a name at the end of calibration
This PA Profile was available directly on my printer when I remove the spool and put back in, but it's not available in BL Studio.  The drop list for my custom filament was empty.

### Actual results

PA Profile not available in BL Studio for custom filament, but available directly on printer

### Expected results

PA Profile available in BL Studio for custom filament

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2023-11-30:
Thanks! we will fix it in the next version.

OzDruiD, 2023-12-07:
@alienm16  have you tried latest beta, Public Beta # 3  this is now working for me (eg drop downs for PA Profiles in Device TAB)
PS- dont know if you noticed, but previouslyt this still worked on  Pritner-AMS scren , just not in slicer.

alienm16, 2023-12-07:
> @alienm16 have you tried latest beta, Public Beta # 3 this is now working for me (eg drop downs for PA Profiles in Device TAB) PS- dont know if you noticed, but previouslyt this still worked on Pritner-AMS scren , just not in slicer.

No, i have not update now.  I will check that soon.

zhimin-zeng-bambulab, 2024-01-18:
The latest version has fixed this problem(e.g. v1.8.4), you can have a try.

closed 2024-01-18 by zhimin-zeng-bambulab

alienm16, 2024-01-18:
yes, it's work fine now.  Thank you