bambulab/BambuStudio #2972: Allow Live Z-axis Adjust on the X1C

mtarallo, 2023-11-26:
We need the ability to adjust the Z-axis. Your auto-calibration is not perfect and I have seen many people complain about the materials either sticking too well or not enough. One of the best printers the Prusa MK3s - and MK4 allow you to adjust the z - this is how you can dial in a perfect first layer. You machine has caused lots of first layer issues. Please give us the ability to adjust the z-axis. Thank you.

sanguinaria-canadensis, 2023-11-28:
there are some limitation the basic motion control algorithm of my motion controller that made the live z adjust is very hard to realize ... 
currently the solution is to use "G29.2 Z{value}" to adjust the z-offset in the start gocde

sanguinaria-canadensis, 2023-11-28:
oooooh, sorry , teh z-adjust cmd is not "G29.2" (this is for turn on/off bed leveling) ,it is "G29.1 Z{value}"