bambulab/BambuStudio #2981: Bambu Studio instant segfault when i try and view the camera

jasonblewis, 2023-11-28:
### Bambu Studio Version

BambuStudio_linux_ubuntu_v01.08.00.62-20231123075101.AppImage 

### Where is the application from?

Bambu Lab github releases

### OS version

Debian 12

### Additional system information

(bambu-studio:252356): Gtk-CRITICAL **: 18:25:09.640: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
Segmentation fault
jason@almond:~/Downloads$ ./BambuStudio_linux_ubuntu_v01.08.00.62-20231123075101.AppImage 

### Printer

Bambu P1S with AMS

### How to reproduce

click on the device tab
click the green play button to view the camera
app segfaults immediatly

```
(bambu-studio:261874): Gtk-CRITICAL **: 18:42:43.227: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
Segmentation fault
jason@almond:~/Downloads$ 
```

### Actual results

app segfaults

```
(bambu-studio:261874): Gtk-CRITICAL **: 18:42:43.227: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
Segmentation fault
jason@almond:~/Downloads$ 
```

### Expected results

app should show me the camera view

### Project file & Debug log uploads

[debug_Tue_Nov_28_18_46_23_265968.log.0.gz](https://github.com/bambulab/BambuStudio/files/13485116/debug_Tue_Nov_28_18_46_23_265968.log.0.gz)

[crashproject.3mf.gz](https://github.com/bambulab/BambuStudio/files/13485128/crashproject.3mf.gz)

### Checklist of files to include

- [X] Log file
- [X] Project file

lanewei120, 2023-12-01:
@jasonblewis 

how about the fedora image?

we have a known issue https://github.com/bambulab/BambuStudio/issues/1814 for jpeg version mismatch
could you also help to check your jpeg version used by gstreamer following https://github.com/bambulab/BambuStudio/issues/1814#issuecomment-1550620932

aduewel, 2023-12-20:
@lanewei120 I am running debian 12, too:

```
readelf -d /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstjpeg.so                                                                                                               1 ↵

Dynamic section at offset 0xcd70 contains 32 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libgstreamer-1.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libgobject-2.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libglib-2.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libgstbase-1.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libgstvideo-1.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libjpeg.so.62]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x000000000000000e (SONAME)             Library soname: [libgstjpeg.so]
 0x000000000000000c (INIT)               0x3000
 0x000000000000000d (FINI)               0x9334
 0x0000000000000019 (INIT_ARRAY)         0xdc90
 0x000000000000001b (INIT_ARRAYSZ)       8 (bytes)
 0x000000000000001a (FINI_ARRAY)         0xdc98
 0x000000000000001c (FINI_ARRAYSZ)       8 (bytes)
 0x000000006ffffef5 (GNU_HASH)           0x260
 0x0000000000000005 (STRTAB)             0xdf8
 0x0000000000000006 (SYMTAB)             0x288
 0x000000000000000a (STRSZ)              2960 (bytes)
 0x000000000000000b (SYMENT)             24 (bytes)
 0x0000000000000003 (PLTGOT)             0xdfe8
 0x0000000000000002 (PLTRELSZ)           2736 (bytes)
 0x0000000000000014 (PLTREL)             RELA
 0x0000000000000017 (JMPREL)             0x1dd8
 0x0000000000000007 (RELA)               0x1af0
 0x0000000000000008 (RELASZ)             744 (bytes)
 0x0000000000000009 (RELAENT)            24 (bytes)
 0x000000006ffffffb (FLAGS_1)            Flags: NODELETE
 0x000000006ffffffe (VERNEED)            0x1a80
 0x000000006fffffff (VERNEEDNUM)         2
 0x000000006ffffff0 (VERSYM)             0x1988
 0x000000006ffffff9 (RELACOUNT)          26
 0x0000000000000000 (NULL)               0x0
```

Starting video with ubuntu image fails:
```
[1]    1253416 segmentation fault  ./BambuStudio_linux_ubuntu_v01.08.02.56-20231213.AppImage
```

Fedora app image works! THANKS!
[image]

ddirector51, 2023-12-24:
I am having the same issue as described by @jasonblewis when the ticket was created.

bambu123, 2023-12-25:
@jasonblewis @ddirector51 @aduewel 
Can you help me call practice run with gdb in the console to get more error messages?

ddirector51, 2023-12-25:
I will try to run with gdb tomorrow or the next day, and let you know. I
haven’t run gdb in years.

aduewel, 2024-01-01:
@bambu123
**short**:
```
gdb BambuStudio_linux_ubuntu_v01.08.02.56-20231228.AppImage
...
Thread 1642 "source:src" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffc8fb7d6c0 (LWP 174053)]
____longjmp_chk () at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:167
167	../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S: No such file or directory.
(gdb) bt
#0  ____longjmp_chk () at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:167
#1  0x834dc2db5d8dbc3c in  ()
```

**Full output:** [gdb_BambuStudio_linux_ubuntu_v01.08.02.56-20231228.AppImage.txt](https://github.com/bambulab/BambuStudio/files/13805480/gdb_BambuStudio_linux_ubuntu_v01.08.02.56-20231228.AppImage.txt)

bambu123, 2024-01-02:
> @bambu123 **short**:
> 
> ```
> gdb BambuStudio_linux_ubuntu_v01.08.02.56-20231228.AppImage
> ...
> Thread 1642 "source:src" received signal SIGSEGV, Segmentation fault.
> [Switching to Thread 0x7ffc8fb7d6c0 (LWP 174053)]
> ____longjmp_chk () at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:167
> 167	../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S: No such file or directory.
> (gdb) bt
> #0  ____longjmp_chk () at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:167
> #1  0x834dc2db5d8dbc3c in  ()
> ```
> 
> **Full output:** [gdb_BambuStudio_linux_ubuntu_v01.08.02.56-20231228.AppImage.txt](https://github.com/bambulab/BambuStudio/files/13805480/gdb_BambuStudio_linux_ubuntu_v01.08.02.56-20231228.AppImage.txt)
> 
> Let me know if you need more information. I am not that experienced with gdb, but I can offer a call/screen sharing session.

It seems to be an inconsistency issue with the JPEG library.
 the same as https://github.com/bambulab/BambuStudio/issues/1814
and https://github.com/bambulab/BambuStudio/issues/1665#issuecomment-1550380137

could you have a check in the above issues?

aduewel, 2024-01-02:
Yes, already did this and also commented there. ;)

Would it be possible to include the jpeglib to the appimage?

lanewei120, 2024-01-16:
how about using this flatpat image 
https://flathub.org/apps/com.bambulab.BambuStudio

lanewei120, 2024-08-08:
> Yes, already did this and also commented there. ;)
> 
> Would it be possible to include the jpeglib to the appimage?

jpeglib is included by gstreamer, which is conflict with the one included in bambustudio

closed 2025-04-29 by MackBambu