bambulab/BambuStudio #2983: Fatal Error afer installation Public Beta 2 on Windows 11

gabber130779, 2023-11-28:
### Bambu Studio Version

1.8.0 public beta 2

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

Intel core I7 9th Gen

### Printer

P1P

### How to reproduce

Install Public Beta 2 on Windows 11

### Actual results

[image]

### Expected results

Bambu Studio should run

### Project file & Debug log uploads

di

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2023-12-01:
@gabber130779 it seems the config files corrupted on your local directory

could you clear these folder and files and retry again
[image]

gabber130779, 2023-12-01:
Okay, that's it. Thanks.

closed 2023-12-01 by gabber130779