bambulab/BambuStudio #2984: 3Dconnexion Wireless Space Mouse no side to side motion

sprior, 2023-11-28:
### Bambu Studio Version

1.8.0 Public beta 2

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

Core i7,
RTX 3070 Ti
32G RAM

### Printer

X1C

### How to reproduce

Start Bambu Studio, select new project, no need to place any models.
Using a Spacemouse move up/down/forward/back and rotate  - all good.
Then try to shift to the left or right and it moves REALLY slow.

### Actual results

With a Spacemouse movement to the left or right is very slow, all other directions seem normal.

### Expected results

I tried this in Super Slicer as a comparison and movement to the left or right seemed normal there.

### Project file & Debug log uploads

[cube.zip](https://github.com/bambulab/BambuStudio/files/13493371/cube.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

bam-winterfell, 2024-01-18:
@sprior May I ask what model of 3D mouse you are using and can you record a video of the problem.
Thank you.

sprior, 2024-01-18:
@bam-winterfell  I've got the Space Mouse Wireless.  I'll try to remember to record a video, haven't tried it in a while.

bam-winterfell, 2024-01-18:
@sprior Sorry for the late reply.I will also try to reproduce this issue.

lanewei120, 2024-10-09:
@Haidiye00 will check it

hagmonk, 2024-12-13:
I think there is already a fix for this issue. `View / Set 3Dconnexion` and you can now adjust the speed of the side to side motion, such that it matches with what you expect. @lanewei120 recommend evaluating this issue as resolved :)

seltzdesign, 2025-09-24:
Just came across this as well. I had to set the translation multiplier to the highest setting (30) to be noticeable. I think the default multiplier of 1 is just wrong and there is some internal multiplier that is not set correctly.