bambulab/BambuStudio #2985: Walls generated inside of the model

loraderon, 2023-11-28:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

Painted models get walls inside instead of infill

[image]

### Actual results

Slow (high quality) walls inside model

### Expected results

No walls inside of models

### Project file & Debug log uploads

[Gnome_n1__Supportless__by_BODY3D-v2.zip](https://github.com/bambulab/BambuStudio/files/13494067/Gnome_n1__Supportless__by_BODY3D-v2.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

lanewei120, 2023-12-01:
@loraderon 

hello, you can find there are different regions on that layer
the wall inside is indeed at the outer of the sub-regions

loraderon, 2023-12-06:
> yes, it is related with multi-color painting

Do you agree that this is an issue and a potential huge perfromance gain in terms of printing times?