bambulab/BambuStudio #2986: Error: No Layers were detected

BahamutEpyon, 2023-11-29:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 

### Printer

Bambu P1P

### How to reproduce

1. Load an stl
2. slice it with default print settings
3. Error: No Layers were detected.
4

### Actual results

It does not slice the models I've been printing without issue for the last month. Changing to a different stl does not change the outcome. Neither does using the fix model function.

### Expected results

It should slice the print.

### Project file & Debug log uploads

[debug_Tue_Nov_28_16_50_50_1832.log.zip](https://github.com/bambulab/BambuStudio/files/13494952/debug_Tue_Nov_28_16_50_50_1832.log.zip)
[MTG_Deck_Box.zip](https://github.com/bambulab/BambuStudio/files/13494954/MTG_Deck_Box.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2023-12-01:
[image]
There are 4 very small objects in this file that cannot be sliced as show in the picture. You can delete these objects and try again.

BahamutEpyon, 2023-12-01:
Problem solved! Thank you for your time.~Kat

 
 

image.png (view on web)