bambulab/BambuStudio #2987: Login Issue

AL2327, 2023-11-29:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Version 10.0.22621 Build 22621

### Additional system information

My full PC specs are in the attached CPU-Z Report. 
[DESKTOP-CPU-Z REPORT.txt](https://github.com/bambulab/BambuStudio/files/13495003/DESKTOP-CPU-Z.REPORT.txt)

### Printer

X1 Carbon

### How to reproduce

1. Start Bambu Studio
2. Wait for Bambu Studio
3. Click Login/Register

### Actual results

[image]

### Expected results

[image]

I have also checked the language preferences and it is set to English. I have even switched it to another language reloaded, back to english, reloaded, and still nothing. 

I have also deleted the EBWebView folder as suggested in another  issue and that did not work. 

I have also tried setting location to different regions, with a VPN, without a VPN, sacrificing my first born to the blood god, and still no luck. 

### Project file & Debug log uploads

[debug_Tue_Nov_28_19_06_13_15644.txt](https://github.com/bambulab/BambuStudio/files/13494981/debug_Tue_Nov_28_19_06_13_15644.txt)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2023-11-29:
The network plugin has not been installed. Please rerun the setup wizard and select Install Plugin in the last step.

AL2327, 2023-11-29:
Thank you for the prompt response. I will uninstall and reinstall bambu studio this evening and report back.

AL2327, 2023-11-30:
I have uninstalled and reinstalled bambu studio. There is no setup wizard. Not at the end of installation, nor does one pop up when launching bambu studio for the first time.

closed 2023-11-30 by AL2327