bambulab/BambuStudio #2990: language help

osmanoney, 2023-11-29:
orca slicerde bu mevcut varken bambu studio türkçe dil desteği ne zaman gelecek ?

walterwongbbl, 2023-11-30:
@lanewei120

lanewei120, 2023-11-30:
@osmanoney 

Hi 
I am afraid we have limited resources
maybe we can wait someone from community to help to add this support

the detailed guide can be found here
https://wiki.bambulab.com/en/bambu-studio/contribute-languages

lanewei120, 2023-12-02:
there are also lots of difference between these two,
we can not use the languange files directly

osmanoney, 2023-12-02:
> bu ikisi arasında da çok fark var, dil dosyalarını doğrudan kullanamıyoruz

Türkiyede 100 lerce bambu lab kullanıcısı var bu konuyuı hızlı bir şekilde çözümlenirse çok sevinirim. ben bile 1 adet le başladım çoğaltmayı düşünüyorum. dil sorunum var yaş 54 çok zorlamnıyorum.

alper38k, 2024-03-27:
Türkiyede bambu çok seviliyor yüzlerce değil binlerce kullanıcısı var acilen dil desteği gelmeli hem cihazlarımıza hemde Bambu lab stuio ya Türkçe dil desteği istiyoruz

lanewei120, 2024-03-27:
just found someone have added it into github
https://github.com/bambulab/BambuStudio/pull/3778

we will follow up

fatih5228, 2024-03-27:
GitHub'a bir çekme isteği gönderdim ve bunun tamamlanmasını bekliyorum. Dün 01.09.00.60 beta sürümüne güncelledim.

fatih5228, 2024-03-27:
@alper38k @osmanoney Biraz daha sabır, gereken Türkçeleştirme yapıldı onay bekliyor. Beta sürüme gelmesini çok isteerdim malesef onaylanmadı. Github da işler malesef biraz yavaş ilerliyor

lanewei120, 2024-03-29:
@StoneLiBambu is processing
it will be in next version

osmanoney, 2024-04-03:
şimdiden emeği geçen herkese şükranlarımı iletiyorum. Bamnu ailesinde olmaktan keyif alıyorum.

closed 2024-06-26 by lanewei120