bambulab/BambuStudio #2991: Sending gcode to a printer doesn't verify if that gcode was sliced for that printer model - you can send invalid gcode

arekm, 2023-11-29:
### Bambu Studio Version

1.8.0.62

### Where is the application from?

Bambu Lab github releases

### OS version

All

### Additional system information

Doesn't matter.

### Printer

X1C

### How to reproduce

1. download gcode file: [wilgotnosc 1.gcode.3mf.gz](https://github.com/bambulab/BambuStudio/files/13505731/wilgotnosc.1.gcode.3mf.gz)
It is a file sliced for A1 printer, so wrong printer.
3. decompress it
4. go to bambu studio, open this file
5. change from "Print" to "Send" to send this file to sd card
6. choose X1C printer from the list
7. click "Send"

### Actual results

File is transferred to the printer without any problem.

What's worse later printer firmware doesn't verify that it was sliced for a different printer and you can start printing from it (with very bad results like head crashing into case etc).

### Expected results

Expected result is Bambu Studio telling me "The selected printer is incompatible with the chosen printer presets". Exactly like it happens when we try to "Print" this file. Whether it should be possible to force sending such file to sdcard or not - that's open for discussion BUT big warning should be there.

### Project file & Debug log uploads

[wilgotnosc 1.gcode.3mf.gz](https://github.com/bambulab/BambuStudio/files/13505731/wilgotnosc.1.gcode.3mf.gz)

### Checklist of files to include

- [ ] Log file
- [X] Project file

walterwongbbl, 2023-12-01:
Hi. @arekm 
Thank you very much, you are absolutely right. We only checked the compatibility between presets and printers during print job submission. We will plan to incorporate this change in the next version.

walterwongbbl, 2023-12-04:
> 在Bambu Studio方面很棒。
> 
> 但固件也应该检查兼容性（如果有人将带有文件切片的 SD 卡放在错误的打印机上并尝试打印它们）。
> 
> 不知道 bambu 打印机是否理解 M862.3 （https://reprap.org/wiki/G-code#M862.3:_Model_name），但这可以用于此。

> @walterwongbbl Great on Bambu Studio side.
> 
> But firmware also should check for compatibility (if someone puts sd card with files sliced for wrong printer and tries to print them).
> 
> No idea if bambu printers understand M862.3 (https://reprap.org/wiki/G-code#M862.3:_Model_name) but that could be used for this.

@sodachen22