bambulab/BambuStudio #2993: 从AMS材料同步列表失败

zlkent, 2023-11-30:
### Bambu Studio Version

1.8.0.62

### Where is the application from?

Bambu Lab github releases

### OS version

Win11 23H2

### Printer

P1SC

### How to reproduce

启动了Bambu Studio后再开启打印机，会导致AMS材料同步失败的情况。
我的操作流程是：在MakerWorld上看到想打的模型，点击“在Bambu Studio中打开”，预览模型的具体参数和效果，确定要打后，再开启打印机电源。在Bambu Studio里切换到设备确认硬件已准备就绪，切换到准备界面，点击AMS同步耗材按钮

### Actual results

同步错误，会丢失部分耗材槽位数据

### Expected results

而在设备开机状态下，启动Bambu Studio并点击同步，则同步正确。

### Project file & Debug log uploads

没有log文件，该操作后log目录下没有当前时间的log文件生成

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zlkent, 2023-12-01:
log已上传
并附上图片
[image]
[image]
[log.zip](https://github.com/bambulab/BambuStudio/files/13524880/log.zip)

该问题在官网提供的1.7版本里也存在。

zlkent, 2023-12-05:
@bambu123 我在测试前，将%APPDATA%\BambuStudioBeta\log目录下的log全清空后测试，测试完成将所有文件打包上传，所以不清楚为什么没包含AMS按钮的相关日志。我又重新复现了一次，不知道这次log是否带有记录。
[log.zip](https://github.com/bambulab/BambuStudio/files/13554502/log.zip)

bambu123, 2023-12-05:
> @bambu123 我在测试前，将%APPDATA%\BambuStudioBeta\log目录下的log全清空后测试，测试完成将所有文件打包上传，所以不清楚为什么没包含AMS按钮的相关日志。我又重新复现了一次，不知道这次log是否带有记录。 [log.zip](https://github.com/bambulab/BambuStudio/files/13554502/log.zip)

好的，谢谢。我看一下新的日志。

bambu123, 2023-12-05:
@zlkent 抱歉还是没有。我能不能远程看看，可以联系我的 QQ 357215492。

zlkent, 2023-12-06:
@bambu123 已发出好友申请，未通过。

bambu123, 2024-02-06:
这里不确定要多久，可以加个“一些材料的信息还不完整，建议稍后再同步一次”