bambulab/BambuStudio #2997: No Build plates

activereality, 2023-11-30:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Mac - ventura 13.2.1

### Printer

X1C

### How to reproduce

1. Open BLStudio
2. Select existing or new Project

### Actual results

No Build plates visible
Cannot see any models - even when placing a new one into a project

### Expected results

See build plates and models

### Project file & Debug log uploads

no log files generated

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2023-12-01:
@activereality 
hello is it a 100% reproducable issue?

is there any logs here  /Users/**user.name**/Library/Application Support/BambuStudio/log?

you can find the guide
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log
to export logs

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95