bambulab/BambuStudio #2998: Support Pattern Angle Not Working

ryantrax, 2023-11-30:
### Bambu Studio Version

1.8.0.62

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

Processor: 12th Gen Intel(R) Core(TM) i7-12700KF   3.60 GHz
Installed RAM: 64.0 GB (63.9 GB usable)
NVIDIA GeForce RTX 3060

### Printer

BambuLab X1C, BambuLab P1S, BambuLab A1 Mini

### How to reproduce

Changing the support pattern angle does not change anything anymore. The support pattern is stuck at a 45 degree angle no matter what angle you change it to
[image]

### Actual results

When changing the angle to 90 degrees (or any other angle) it remains 45 degrees
[image]

### Expected results

The support pattern angle should match the angle imputed. 

### Project file & Debug log uploads

Not required.

### Checklist of files to include

- [X] Log file
- [X] Project file

tangketan, 2023-12-01:
This bug has been fixed. Please wait for the official v1.8 release.

tangketan, 2023-12-21:
Offical 1.8 has been released. Close this issue.

closed 2023-12-21 by tangketan