bambulab/BambuStudio #2999: Video streaming on Windows 10 and VLC Media Player Freezes after first frame

Rich-b1, 2023-11-30:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

VLC Media Player version 3.0.20
OBS Studio version 30.0.0

### Printer

Bambu Lab X1C

### How to reproduce

Install video streaming support via the "Show Live Streaming Guide Page" in Bambu Studio.

Verified that the live streaming works properly with OBS Studio on Windows 10. Close OBS Studio.

Run VLC Player, using the file ffmpeg.sdp as a video source
Notice that the first video frame from the X1C is displayed, but then the view track time slider is wandering back and forth. Also the video is frozen at the first frame.

Close VLS Player, then launch OBS Studio again to confirm that the video is working ok.

### Actual results

VLC Media Player will display only one frame of video from the X1C.

However, OBS Studio will work correctly at all times.

### Expected results

VLC Media player should be able to display the running video from the X1C

### Project file & Debug log uploads

No project, simply run VLC Media Player as described.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

OzDruiD, 2023-12-01:
> You can use ffplay.
> D:\Tools\ffplay.exe C:\Users\Administrator\AppData\Roaming\BambuStudioInternal\cameratools\ffmpeg.sdp -protocol_whitelist file,udp,rtp

This hardly ever works for me, I run the one from the  Beta\CameraTools folder, but with the same paramaters..

[image]

Rich-b1, 2023-12-03:
I Installed ffmpeg (ffplay) and also didn't have much luck! I also tried it running as an admin, but no change.

I didn't want to debug/analyze/engineer this myself, but simply report this to BambuLab to consider it a bug to be fixed as it's on the Wiki as if it should be working.

Here's the Wiki page i followed:
https://wiki.bambulab.com/en/software/bambu-studio/virtual-camera

Here's the error output ffplay,  I'm not sure if this is related to the reason why VLC Media Player is not working as this is the first time I've tried ffplay:

[sdp @ 000002a53d3c09c0] Could not find codec parameters for stream 0 (Video: h264, none): unspecified size
Consider increasing the value for the 'analyzeduration' (0) and 'probesize' (5000000) options
Input #0, sdp, from 'C:\Users\Rich\AppData\Roaming\BambuStudio\cameratools\ffmpeg.sdp':
  Metadata:
    title           : No Name
  Duration: N/A, bitrate: N/A
  Stream #0:0: Video: h264, none, 90k tbr, 90k tbn
    nan M-V:    nan fd=   0 aq=    0KB vq=    0KB sq=    0B f=0/0

Rich-b1, 2023-12-03:
This is not working at all with VLC Player, nor ffplay

bambu123, 2023-12-04:
Is your sdp files correct?

v=0
o=- 0 0 IN IP4 127.0.0.1
s=No Name
c=IN IP4 127.0.0.1
t=0 0
a=tool:libavformat LIBAVFORMAT_VERSION
m=video 1234 RTP/AVP 96
a=rtpmap:96 H264/90000
a=fmtp:96 packetization-mode=1; sprop-parameter-sets=Z0LAH42NUCSC2TZAAAADAEAAAA8jwiEagA==,aM4xsgA=; profile-level-id=42C01F

Rich-b1, 2023-12-06:
Yes, These are the same as located in my ffmpeg.sdp file as created by Bambu Studio.

I haven't touched it at all. In fact this is the first time that I even looked inside of this file.

I don't know if these are correct, but they match what my file has in it.

Does VLC Media player work for you in Windows 10?

Rich-b1, 2023-12-06:
I'm not sure what changed. All I did was open the ffmpeg.sdp file with my text editor, and then close it. (I did NOT save the file, only view it).

Now VLC Player is working!  This is strange! I don't know what changed.
I will say that just prior to testing this, I had changed my complete hot end. So the power to my printer was turned off, and then on again.

Please ignore this issue. I'll close it for now. And only re-open it again if I can find the conditions to cause it to re-occur on demand.

Sorry for the waste of time!

Rich-b1, 2023-12-12:
I feel like I'm on a roller coaster ride with this issue! LOL

It seems to be working and more stable now with firmware 1.07.00.00 on my X1C.

I'll close this issue and only reopen it if it starts again.

Thank you.

closed 2023-12-12 by Rich-b1

Wise-By-Design, 2026-09-05:
I have been having this very same issue on an H2D. I'm not sure what fixes it, but I can get it working sometimes, basically doing all the same things that you have mentioned. I am currently at the point of it not working again and have not been able to get it working again yet.

Noggin01, 2026-09-11:
**Edit:** _I'm on an H2D printer, so this may not apply to others. I'll open a separate bug report with this information later today._

Go Live feature enables a camera stream, and what we're seeing is that VLC chokes on it and shows a single frame. If you are in VLC and press the fast-forward button, you'll get a new frame. If you get the new frame, that indicates that the printer/software is streaming, but the stream is malformed.

_All commands indicated below are run from **C:\Users\[username]\AppData\Roaming\BambuStudio\cameratools>**_

Run this command; **ffmpeg -protocol_whitelist file,udp,rtp -i ffmpeg.sdp -f null NUL**

I get a bunch of errors, then about a 5 second delay, then something that looks like legitimate output. The final line of this output is:

_frame= 1252 fps= 33 q=-0.0 size=N/A time=00:00:00.03 bitrate=N/A speed=0.00088x_

The data is being streamed at 30 frames per second, or 0.0333 seconds per frame. Frame 1252 should be at a timestamp of 41.73 seconds, but it and every single other frame has a timestamp of 0.03 seconds. The timestamps embedded in the stream are _not advancing_ so VLC chokes on it.

Further support of this being the issue is **ffmpeg -protocol_whitelist file,udp,rtp -i ffmpeg.sdp -vf "setpts=N/(30*TB)" -f nut - | ffplay -**

This generates a line of text for every received frame from the stream, and the PTS and DTS fields are all 0.

The following command opens the stream in a video player and uses your computer's clock to generate a timestamp for the stream so that it actually plays: **ffplay -protocol_whitelist file,udp,rtp -use_wallclock_as_timestamps 1 -vf "setpts=PTS-STARTPTS" -framedrop -sync ext ffmpeg.sdp**

Rich-b1, 2026-10-11:
Noggin01,

Thank you for your work on this!

I wonder if the root problem is in the device (X1C , H2D, etc.) firmware or the Bambu Studio software? 

I havn't been using streaming in a long time, so I'm not sure if this is still an issue for the X1C.

It sure would be nice if Bambu could fix this for all printers!

Wise-By-Design, 2026-10-12:
> 19\. **Edit:** _I'm on an H2D printer, so this may not apply to others. I'll open a separate bug report with this information later today._
>      **Edit 2:** _[#12201](https://github.com/bambulab/BambuStudio/issues/12201)_
>      Go Live feature enables a camera stream, and what we're seeing is that VLC chokes on it and shows a single frame. If you are in VLC and press the fast-forward button, you'll get a new frame. If you get the new frame, that indicates that the printer/software is streaming, but the stream is malformed.
>      _All commands indicated below are run from **C:\Users[username]\AppData\Roaming\BambuStudio\cameratools>**_
>      Run this command; **ffmpeg -protocol_whitelist file,udp,rtp -i ffmpeg.sdp -f null NUL**
>      I get a bunch of errors, then about a 5 second delay, then something that looks like legitimate output. The final line of this output is:
>      _frame= 1252 fps= 33 q=-0.0 size=N/A time=00:00:00.03 bitrate=N/A speed=0.00088x_
>      The data is being streamed at 30 frames per second, or 0.0333 seconds per frame. Frame 1252 should be at a timestamp of 41.73 seconds, but it and every single other frame has a timestamp of 0.03 seconds. The timestamps embedded in the stream are _not advancing_ so VLC chokes on it.
>      Further support of this being the issue is **ffmpeg -protocol_whitelist file,udp,rtp -i ffmpeg.sdp -vf "setpts=N/(30*TB)" -f nut - | ffplay -**
>      This generates a line of text for every received frame from the stream, and the PTS and DTS fields are all 0.
>      The following command opens the stream in a video player and uses your computer's clock to generate a timestamp for the stream so that it actually plays: **ffplay -protocol_whitelist file,udp,rtp -use_wallclock_as_timestamps 1 -vf "setpts=PTS-STARTPTS" -framedrop -sync ext ffmpeg.sdp**

Can confirm, I started digging around and I have done similar kind of testing with the same results. This needs looked into and fixed among the other networking discrepancies. And can we get access to the other cameras? Please?