bambulab/BambuStudio #3001: Issue with opening makerworld website “Failed to open URI”

Xue-JW, 2023-11-30:
### Bambu Studio Version

1.8.0 beta2

### Where is the application from?

Bambu Lab github releases

### OS version

debian12

### Printer

P1S

### How to reproduce

click import to Bambu Studio

### Actual results

Failed to open URI.
The specified location is not supported.

bambustudio://open?file=xxxxxxx.3mf

### Expected results

open 3mf in Bambu Studio

### Project file & Debug log uploads

temporary solution：
1. execute Bambu Studio.AppImage
2. copy "BambuStudio.png" and "BambuStudio.desktop" in the hidden folder `/tmp/mount_Bambu*` to user home
3. copy BambuStudio.desktop to /usr/share/applications with sudo
4. add %U at the end of Exec, For example:
/usr/share/applications/BambuStudio.desktop
```
[Desktop Entry]
Name=BambuStudio
Exec=/home/gavinx/app/bambu/BambuStudio_linux_ubuntu_v01.08.00.62-20231123075101.AppImage %U
Icon=/home/gavinx/app/bambu/BambuStudio.png
Type=Application
Categories=Utility;
MimeType=model/stl;application/vnd.ms-3mfdocument;application/prs.wavefront-obj;application/x-amf;
```
5.  add `x-scheme-handler/bambustudio=BambuStudio.desktop;` at the end of /usr/share/applications/defaults.list

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-12-18:
@Xue-JW 
Sorry, we do not yet support opening studio directly from Makerworld on Linux. But when there is ample time, we will add this feature.

r14n, 2024-04-17:
Any progress on this or did someone figure it out?

r14n, 2024-04-17:
I was able to right click the 3mf in Chrome downloads and select always open.  So now the download link works as the open in bambu studio link I guess, unless that link has even more features or integration than opening the 3mf.

b1ggi, 2024-09-23:
Should still be investigated

token47, 2024-10-17:
This works in Linux if configured manually. The secret is the line in defaults.list file (which can also be in your home dir, does not need to be the system one). Thank you for sharing.