bambulab/BambuStudio #3004: Pointless slim trim support

TennisRunner, 2023-11-30:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu p1p

### How to reproduce

0.1mm layer
type: tree(auto)
Style: Tree Slim

### Actual results

Creates a support tree that clips into the side wall, providing no support
[image]

### Expected results

Should not make that support at all or make it go around the outer wall.

### Project file & Debug log uploads

[pointless-tree-support.zip](https://github.com/bambulab/BambuStudio/files/13517812/pointless-tree-support.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tangketan, 2023-12-01:
It's a known issue, not easy to fix. The latest v1.8 is slightly better. We'll continue to improve.
[image]

TennisRunner, 2023-12-01:
> It's a known issue, not easy to fix. The latest v1.8 is slightly better. We'll continue to improve. [...]

Alrighty