bambulab/BambuStudio #3006: Allow running Post-processing scripts when using printing to printer using Print Plate

gary-wlg, 2023-12-01:
It appears that currently the Post-processing scripts are only executed if you export the gcode using File -> Export  -> Export Gcode.    This means there is no way to slice the file and send it to the printer if you want your post processing script(s) to run.

Please add support to allow post-processing scripts to run when sending prints through the cloud to the printer.

bam-winterfell, 2024-03-20:
hi @gary-wlg 
We have noticed such issues and will fix them in the future

ansonl, 2024-11-15:
@bam-winterfell Has there been any progress on using Post-processed g-code for the Send and Print features? It would be great for multi color prints that have been post processed and need AMS slot remapping before print.

ansonl, 2024-12-26:
In the latest version of Bambu Studio 1.10.1.50, you can now Print exported gcode directly from Bambu Studio to the printer but AMS mapping does not work. 

If I post process the exported gcode separately and drag it back into Bambu Studio for preview the print and send buttons are enabled. You can't remap the filaments in the AMS unfortunately (which should be a trivial improvement for Bambu to add) but the gcode file will be sent to the printer and executed.