bambulab/BambuStudio #3009: menu device hms every white

magicozero, 2023-12-01:
### Bambu Studio Version

last official and last beta

### Where is the application from?

Bambu Lab github releases

### OS version

windows 11

### Additional system information

microsoft surface pro 7 

### Printer

bambu lab x1 carbon

### How to reproduce

launch app bambu studio
device menu 
hms menu
white no see nothing

### Actual results

[image]

### Expected results

no see nothing other function is ok and bambu handy on cellular ams work

### Project file & Debug log uploads

[debug_network_Thu_Nov_30_17_15_24.log](https://github.com/bambulab/BambuStudio/files/13522197/debug_network_Thu_Nov_30_17_15_24.log)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bam-winterfell, 2023-12-04:
hi @magicozero 
The HMS page only displays the printer's prompts on the studio. If there is no prompt message from the printer, the HMS page will be blank

bam-winterfell, 2024-01-18:
@magicozero If you have any other questions, please open a new issue.

closed 2024-01-18 by bam-winterfell