bambulab/BambuStudio #3011: Bambu Slicer not slicing correctly

cjschwartz, 2023-12-01:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 

### Additional system information

Intel(R) Core(TM) i7-7700HQ CPU @ 2.80GHz   2.81 GHz

### Printer

X1C

### How to reproduce

Slice the object after preview and watch the white disappear. 

### Actual results

When slicing this file, the white disappears. I have tried this on multiple different files that I created myself in Fusion 360 and exported as 3MF to Bambu slicer but the white only shows up at first and disappears when sliced. 

### Expected results

All colors are to be sliced and included in the print as usual. I should see the object in my preview all get sliced. 

### Project file & Debug log uploads

[WSU File Fixed but not working.zip](https://github.com/bambulab/BambuStudio/files/13528062/WSU.File.Fixed.but.not.working.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2023-12-19:
@cjschwartz  Did the above method solve your problem?

cjschwartz, 2023-12-27:
Yes sir thank you

closed 2024-01-03 by zhimin-zeng-bambulab