bambulab/BambuStudio #3012: Load of 3mf file from FreeCAD 0.21 fails with error "no geometry"

tomunger, 2023-12-01:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

mac os 14.1.1

### Additional system information

  Model Name:	MacBook Pro
  Model Identifier:	MacBookPro18,1
  Model Number:	Z14W00108LL/A
  Chip:	Apple M1 Pro
  Total Number of Cores:	10 (8 performance and 2 efficiency)
  Memory:	32 GB
  System Firmware Version:	10151.41.12
  OS Loader Version:	10151.41.12
  Serial Number (system):	Y204K36JTH
  Hardware UUID:	0D9E2079-39C4-531E-9B25-112185972DE5
  Provisioning UDID:	00006000-001238C41A38801E
  Activation Lock Status:	Enabled

### Printer

X1C

### How to reproduce

1. Open Bambu Studio
2. Open the attached 3mf file
3. Load fails with error "The file does not contain any geometry data"

### Actual results

Most of my 3mf exports from FreeCAD 0.21 fail.  This file represents a very simple object created with a sketch and single pad.  
I can open the 3mf file with FreeCAD and it shows the expected geometry.  
I can export to STL and load those into a Bambu Studio project. 
Exports from FreeCAD 0.20 to 3mf never failed in this way.  It may be some thing new about version 0.21

### Expected results

It should open the file and display the objects.

### Project file & Debug log uploads

[liftmaster_remote_clip.3mf.zip](https://github.com/bambulab/BambuStudio/files/13530167/liftmaster_remote_clip.3mf.zip)
[bambu studio log files.zip](https://github.com/bambulab/BambuStudio/files/13530174/bambu.studio.log.files.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

michalxo, 2023-12-05:
Hello, I have experienced the very same problem. First it opened 3mf file without problem. Then after more modifications, it started to shout this bug. But it's the same file. 
[biddingbox3.3mf.zip](https://github.com/bambulab/BambuStudio/files/13558282/biddingbox3.3mf.zip)
In attached zip are 2 files from same project. Older version works fine, newer is broken.

Both created in FreeCAD 21 & latest BambuStudio 1.8.0.62

Thank you for looking at this.

DanBao-Bambu, 2023-12-05:
@michalxo Thank you for providing the 3mf file, which has given me more information. I don't know if you know that a 3mf file is a model file saved in a certain format, and its model information is saved in "./3D/3dmodel. model". At least the file "biddingbox3.3mf" is described as containing two objects, but their types are saved as "surface" as follows:
[image]
That's why Bambu Studio is unable to parse the model.
I'm not sure if any of your settings have been changed in FreeCAD, or if the default type in the new version of FreeCAD has been replaced with "surface". Please check.

When I manually change the surface to model, it can be recognized and loaded by BambuStudio.

Schnulli87, 2023-12-20:
I have experienced the same behavior with Bambu Studio 1.8.2.56 and FreeCAD 0.21.1.
After reading the comments on this issue, I did some tests on my system, as I have been using the .3mf export for a long time with many versions of Cura and FreeCAD and never experienced anything similar:
I tried to load a model, that I had already previously exported and sliced in Cura, in Bambu Studio which gave me the error message from the first post. I loaded the same file in Cura to confirm that the file is technically ok.
Then, I selected **the part** in FreeCAD and exported it again. Again, it failed loading in Bambu Studio, while working in Cura.
Now, I selected **the last modification of this part** in FreeCAD and exported it. Now, Bambu Studio complained, that the design is from another program, **but loaded the model just fine**. After switching to Cura and clicking "reload changed file", the part was suddenly rotated 90°.
It seems, FreeCAD does a different Export, based on the selection in the model tree.
I confirmed this behavior with several other components from my designed device.
**If a model is exported to .3mf from FreeCAD and the "Part" is selected in the component tree, then Bambu Studio will often not recognize the design. Instead, the last design-step needs to be selected when exporting the model.**

[image]

I could not find out, when exactly FreeCAD changes its export behavior, as I had a simple design (extruded sketch), that showed this behavior after adding a second extrusion on its surface. But a quickly sketched part for this text (extruded sketch and an additional extrusion on its surface) worked all the time in Bambu Studio. Also, after restarting FreeCAD and exporting the invisible part, it could be loaded in Bambu Studio.

**To sum up:**
There might be a difference (bug? feature?) in the export function of FreeCAD, but as the model is always completely ok and can be loaded in other slicer SW (only Cura tested), it should be fixed in Bambu Studio to allow loading of "both" exported models.

Both .3mf versions of the first design from my screenshot are attached (red & green selection).
[Clip_ok_not_ok.zip](https://github.com/bambulab/BambuStudio/files/13732777/Clip_ok_not_ok.zip)

waynger, 2024-01-10:
Solidworks 2020, Bambu Lab 1.8.2.56 win 10 same thing "no geometry" when loading .3mf, however doing a "save as" in sw will work fine in bl even though it is the same file just different filename

tokamac, 2024-02-17:
I have been experiencing the same issue lately, exporting solid models from Shapr3D as STEP files and failing at importing them in latest BambuStudio (1.8.4.51): 

[image]

Choosing various STEP version options when exporting (i.e. AP203, AP242 or default AP214) does not solve the issue. This does not occur with all STEP files exported from Shapr3D. 

I have checked that the model and directories names in the file hierarchy are short and simple without any special character.
The weird thing is that it seems to occur randomly with exported parts from the same Shapr3D project.

The issue is very recent. Exporting the same problematic model as a mesh STL file solves the problem, but I prefer to use STEP files instead.

@DanBao-Bambu, I have emailed you one of the problematic STEP files so you can investigate this issue further.

tomunger, 2024-02-20:
In freeCAD 0.22 (development release) this is no longer a problem.  I'm able to export to 3MF files and Bambu Studio loads them.

tokamac, 2024-02-20:
I suspect that your issue deals with whether or not the files inside your 3MF exported from FreeCAD are mesh-based (like STLs) or solid-based (like STEP files).
The 3MF file format was developed  in the first place as a mesh-based definition for 3D CAD models (like STL). 
But it also acts as a "container" insofar as it can also include non-mesh models. It all depends on the export settings.

Bambu Lab has offered the possibility to import STEP files (solid-based) in BambuStudio. As it happens, the recording format for BambuStudio project files is 3MF. This proves that such file format accepts either mesh-based STL files or solid-based STEP files.

The issue persists with some STEP files.

I prefer to use STEP instead of 3MF or STL whenever I can, because solid-based CAD models are smoother by definition, they are not triangulated before importation. To print circles and arcs, this rocks in conjunction with the setting "Arc fitting" in the slicer. 

So I'd like Bambu to investigate the the import issue you raised, but specifically concerning STEP files.

DanBao-Bambu, 2024-08-08:
@tokamac I'm very sorry for losing this issue. If you are still concerned about this issue, you can create a new one and attach the problematic step file. If not, I will reproduce the problem you mentioned locally. Thank you for providing the information,

dtgolder, 2024-11-08:
This is still an issue (20241108)...happened to me today (but for some models I created last week, everything seemed to work fine).

What solved it for me is the check the "Use legacy export function" when exporting to a STEP file. The legacy import loaded correctly into BambuStudio.

My assumption (perhaps incorrect) is that something changed in the FreeCAD export...

(Note: using FreeCad 0.21.2)

Enclosed sample file that does not export unless the legacy export box is checked...
[Gasket Flange Example.tar.gz](https://github.com/user-attachments/files/17683931/Gasket.Flange.Example.tar.gz)

bcwhite-code, 2024-11-24:
I came here searching the same problem and was about to add my own info... when I realized that the _body_ of my creation was not inside the _part_ and so the part geometry really was empty.  Mentioning this just in case someone else ends up here.

wohltat, 2025-05-03:
I can replicate that error with Orcaslicer 2.3.0 and a FreeCAD version listed below.

I have a part design model that breaks at a certain feature. If i compare the exported .3mf there is a difference in the `3dmodel.model`file inside the 3mf package.

The working file looks like this
```
<?xml version="1.0" encoding="UTF-8"?>
<model unit="millimeter" xml:lang="en-US" xmlns="http://schemas.microsoft.com/3dmanufacturing/core/2015/02">
 <metadata name="Application">FreeCAD</metadata>
 <resources>
  <object id="1" type="model">
...
```
and the broken file like this:
```
...
<object id="1" type="surface">
...
```
I don't know why it changes from `model` to `surface`. It seems like a bug in FreeCAD since i can sometimes export it with a special order of refreshes, but Orcaslicer could also be more tolerant with the data.

--------------------

```
OS: Manjaro Linux (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 2024.1006.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: a9810d509a6f112b5ac03d4d4831b67e6bffd5b7
Python 3.11.9, Qt 5.15.13, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

closed 2026-09-04 by tomunger