bambulab/BambuStudio #3014: Cannot Log In, blank screen & severe window lag

RastonWarrior, 2023-12-02:
### Bambu Studio Version

ALL, incl. v1.7 to new beta 1.8

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 10 Pro 22H2

### Additional system information

CPU : 11th Gen Intel(R) Core(TM) i9-11900K @ 3.50GHz   3.50 GHz
RAM : 64Gb
Storage - 780Gb free
Video : NVidia 2080ti (latest drivers, 12/1/23)

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. bare metal re-install of Windows 10 Pro with all updates
2. install NVidia driver, latest version
3. download and install any version of Bambu Lab, 1.6 or greater as any user, incl. Administrator
4. install to default path
5. after Install Success message, Run Bambu Studio
6. screenshot 1 - blank setup wizard, no keys or mouseclicks within the window do anything except Escape
7. Press Esc - screenshot 2, blank window
8. screenshot 3 - File - Preferences - see that English is selected by default (common fix for this problem, this is not that)
9. setting login region to North America has no affect at any time
10. enable Dark Mode (my preference)
11. Back to base window, click on Device - notice that Network Plugin is not installed, Click here to install it.  Installation is successful (it seems), but it changes nothing
12. screenshot 4 - blank window is home
13. Select File - Help - Open Network Test.  All tests are successful except HTTP: Please Log In First, which I cannot (this is the problem)
14. screenshot 5 - no printer, no device, no login: no functionality
15. Lastly, see windowlagdemo.mp4: at all times, the application window is LAGGY. Everything else on the desktop is great, the machine works just fine, but when you click and drag this application window around it's like you're on a 386 with an old-school video card. This video shows click - drag in one circle - release click and continue moving mouse while the window is still moving slowly in the background.

Application is entirely unusable.
[image]
[image]
[image]
[image]
[image]

https://github.com/bambulab/BambuStudio/assets/16449082/1a4cb322-4e57-48c8-9223-3d254ab78b3a

### Actual results

Regardless of install / uninstall / reinstall  or video drivers or version, application is unusable.  Help - Setup Wizard remains blank.

### Expected results

I've been using this application for 9 months prior to a full system reload - new hard drive, same hardware, reinstalled same OS, but this time around the application is unusable. Everything else on the machine works perfectly.

No project file relevant - can't get that far.
Printer not relevant (X1 Carbon) - can't log in to connect
Again, Help menu network tests are good.

### Project file & Debug log uploads

project file not relevant, can't get that far, debug log follows:

[debug_Fri_Dec_01_19_56_28_11860.log](https://github.com/bambulab/BambuStudio/files/13533221/debug_Fri_Dec_01_19_56_28_11860.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

RastonWarrior, 2023-12-02:
Please note, I submitted a support ticket to Bambu Lab via the mobile app a week ago - no response.

bambu123, 2023-12-04:
Another issue is same as #934

RastonWarrior, 2023-12-04:
It did, plus the Setup Wizard started working. It's like the software just "woke up" and started to do what it was supposed to.  I did nothing, literally closed this GitHub browser window and went to bed. By morning I started Bambu Studio to try again and it just worked.  Clearly something was changed on the back-end.

My only problem now is indeed the same as #934  .

Closing this bug report in deference to existing ticket #934 ,

  - R

closed 2023-12-04 by RastonWarrior