bambulab/BambuStudio #3015: Failed to send the print job

JohnyTapia, 2023-12-02:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu lab X1 Carbon

### How to reproduce

When I want to send something to print from bambu studio, it gets stuck at 30% and after a while it says "Failed to send the print Job. Please try again." Error code: 3030
Error desc: Check the current status of the bambu server
Extra info: http code 28 Timeout was reached....

### Actual results

"Operational" shows your servers on the page but the cloud simply does not respond

### Expected results

well it should start printing

### Project file & Debug log uploads

[debug_network_Sat_Dec_02_21_53_26.log](https://github.com/bambulab/BambuStudio/files/13537425/debug_network_Sat_Dec_02_21_53_26.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

JohnyTapia, 2023-12-04:
I chose the region Europe, i am from Slovakia in europe..

walterwongbbl, 2023-12-05:
Hi. @JohnyTapia 
error: curl:Couldn't resolve host name:Could not resolve host: s3.us-west-1.amazonaws.com [Error 6], body = , status = 6

This error message indicates that the hostname "s3.us-west-1.amazonaws.com" cannot be resolved, resulting in a failed upload. This could be due to network connectivity issues or DNS resolution errors. Please ensure that your network connection is functioning properly and your computer can correctly resolve the hostname for Amazon S3. You can try using a different network or check your DNS settings to resolve this issue.

walterwongbbl, 2023-12-05:
Hi. @gspitman 
If you can provide your log file, I can help you analyze the error.

JohnyTapia, 2023-12-05:
I also tried disconnecting the network and reconnecting the printer to the network, as well as the account, how do I change the host on the printer? I don't understand why it started showing me this, because a few days ago it was working perfectly normally, my internet is working normally

gspitman, 2023-12-05:
In my case, it worked when I turned off my Wi-Fi and I went to Cellular, is
it possible that my network thought it was looping back?

walterwongbbl, 2023-12-06:
Hi.  @gspitman   @JohnyTapia 
If you are unsure of the cause for the software and printer not functioning properly in your network, it could be due to a specific setting or other software changes on your computer that have altered a network property. You can try resetting your network and then attempting again.

You can replace the cellular network like @gspitman  and verify if it is affected by WiFi.

MarekUhrinec, 2023-12-17:
@JohnyTapia  hi, i have some same issue. if i am on my home wifi i can send print job trough bambu handy but if i want to send it trough bambu studio i dont send and it show error 3030. 
i am from slovakia so that why i am asking you.

thanks for your answer

JohnyTapia, 2023-12-20:
@MarekUhrinec  I still have the problem, I don't know where to look for the error, but probably with the provider, because when I tried to create a personal hotspot, it works

walterwongbbl, 2023-12-21:
@MarekUhrinec @JohnyTapia 
It depends on whether your network can access Amazon Web Services (AWS). If you can access AWS using your mobile hotspot but not through your Wi-Fi, you can try checking if there are any firewall or router settings blocking the access. If the issue persists, please contact your local Internet service provider to confirm if they are blocking access to AWS. You can also check the following interface to confirm if Bambu Studio supports various network services.
[image]

turbocomp, 2024-02-03:
I have a problem sending a task when I insert an SD card. Without a memory card, uploading to the cloud is fast, I tested the network parameters and everything was fine. The problem is I can’t find a memory card, I tried different manufacturers, different speeds, I format all cards FAT32, I format the larger 512Gb with exFAT. When you insert a memory card, it is formatted by the printer and even writes some files to it, but after that it does not write anything to the card and writes about the elapsed time.
How to solve this problem?
[image]
[image]
[image]

MarekUhrinec, 2024-02-25:
@JohnyTapia hey i would like to ask what wifi provider you have, because i still dont know where the problem is and the hotspot dont work too, so if it is the wifi provider or its bambu. 

thanks for your answer

JohnyTapia, 2024-02-26:
@MarekUhrinec typujem že si slovak že, no ja mám orange a problém som stále nevyriešil ale keď si spravím hotspot na 5G tak to normalne fachčí, ešte môže byť problém v notebooku len to som zatial nemal ako vyskúšať keďže nový nemám a reinstall win sa mi moc nechce. Zatial fičím na sd kartách aspoň mám všetko naslicovane a uložené pre prípad že by som to niekedy ešte chcel tlačiť. Každopádne ma to hnevá

I'm guessing you're Slovak, right? Well, I have orange and I still haven't solved the problem, but when I make a hotspot on 5G, it's normal, it might be a problem in the laptop, but I haven't had a chance to test it yet, because I don't have a new one and I don't really want to reinstall win . For now, I'm working on sd cards, at least I have everything sorted and saved in case I ever want to print it again. Anyway, it makes me angry

MarekUhrinec, 2024-02-27:
čauko, ja mam tiež orange a myslím si že v orange je problém pretože ja mam aj orange číslo a už ani hotspot mi nefunguje.

Hi, I also have orange and I think that orange is the problem because I also have an orange number and even the hotspot doesn't work anymore.

walterwongbbl, 2024-03-11:
@MarekUhrinec 
I'm very sorry, my translation software seems to be unable to translate your conversation. Has it been confirmed that it is a network issue now.

Reischkan, 2024-05-09:
@walterwongbbl i have a similar problem but wen i try the network test the storage and storage upgrade failed. please help 
[image]

JohnyTapia, 2024-05-09:
I changed the pc and everything works for me now

codekansas, 2024-05-11:
BTW I experienced this issue because I was connected to a VPN - disconnecting fixed it

closed 2024-08-08 by walterwongbbl