bambulab/BambuStudio #3017: Bambu Studio crashing when pressing device tab after loading software

kr320, 2023-12-03:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab X1C

### How to reproduce

Open Bambu Studio
Click on the device tab

### Actual results

The device tab begins to open. Only the left side bar, camera and print progress box loads. 
The mouse pointer turns to Working in Background(pointer with circle). 
About a second later the software closes with no error message.

Note this happens whether you are simply opening bambu studio and clicking device or opening a project. It has happened for about 3 days.

If I click on a few of the other tabs first(Prepare, Preview, Project, Calibration) and then click Device, it opens properly and doesn't crash.

[image]

https://github.com/bambulab/BambuStudio/assets/118021697/39783a11-dc9a-4d67-910b-6f4aae5529ec

### Expected results

The device tab should open and display all information without crashing.

### Project file & Debug log uploads

[Crash.zip](https://github.com/bambulab/BambuStudio/files/13537802/Crash.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

lanewei120, 2023-12-04:
hello @kr320 

is there any dmp files under your log directory?
if yes
could you help to upload here
[image]

kr320, 2023-12-10:
I'm currently unable to duplicate the issue, seems to have resolved itself.

> and is it a 100% reproducible issue on your side?
At the time it was happening, for about 3-4 days, it was 100% reproducible and would happen every time.

> also how about previous versions?
Didn't try previous versions.

Would you still like me to try and see if there is a dmp file?

lanewei120, 2023-12-11:
> I'm currently unable to duplicate the issue, seems to have resolved itself.
>
>
> At the time it was happening, for about 3-4 days, it was 100% reproducible and would happen every time.
>
>
> Didn't try [...]

@kr320 
thanks for the reply
please help to check the dmp file also, maybe we can find some glue for previous crash

kr320, 2023-12-12:
Im running into this problem again.

Same exact issue from the original report. It does it if you simply open the app or if you open a project. After clicking device, Bambu Studio closes.

Series of events:
Opened a project.
Started the print.
Saved the project.
Watched the video for a while. Saved and closed bambu studio.
About 30 minutes later I went to open the project to look at the camera. When I clicked on the device tab, Bambu Studio closes..

I did it a bunch of times. I zipped the 3mf project file and a few logs, dmps and other files from the Log folder at around the time it was happening.

[Crash when clicking Device tab.zip](https://github.com/bambulab/BambuStudio/files/13642831/Crash.when.clicking.Device.tab.zip)

bambu123, 2023-12-12:
Exception Code :c0150010 The activation context being deactivated is not active for the current thread of execution.

> Im running into this problem again.
>
> Same exact issue from the original report. It does it if you simply open the app or if you open a project. After clicking device, Bambu Studio closes.
>
> Series of [...]

To analyze DMP files, we need you to provide the following DLLs on your system:

C:\Windows\System32\ntdll.dll
C:\Windows\System32\kernel32.dll
C:\Windows\System32\KERNELBASE.dll
C:\Windows\System32\user32.dll
C:\Windows\System32\msvcp140.dll
C:\Windows\System32\ucrtbase.dll
C:\Windows\System32\win32u.dll

kr320, 2023-12-14:
[DLLs.zip](https://github.com/bambulab/BambuStudio/files/13668320/DLLs.zip)

bambu123, 2023-12-14:
The DLL (bdhkm64.dll) injected by other software may have caused the problem.

[What is bdhkm64.dll?](https://www.freefixer.com/library/file/bdhkm64.dll-311612/)
bdhkm64.dll is part of BitDefender® AntiVirus and developed by BitDefender S.R.L. Bucharest, ROMANIA according to the bdhkm64.dll version information. bdhkm64.

kr320, 2023-12-14:
Any suggestions? I assume I can't delete this since its part of my anti virus software. I'll contact BitDefender and see what they say

bambu123, 2024-08-08:
Is there a reply from BitDefender ?

closed 2024-10-17 by walterwongbbl