bambulab/BambuStudio #3018: macOS - Can't connect to any printers when a "new window" is open.

npike, 2023-12-03:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 10.15.7

### Printer

Bambu Lab P1S and P1P

### How to reproduce

1. Open first instance of Bambu Studio.  Verify Device tab shows all LAN printers available (a P1S and a P1P)
2. Right click on Bambu Studio in dock, select "New Window"
3. Click on "Device" tab

When you close the 2nd instance of Bambu Studio, the network log in ~/Library/Application Support/BambuStudio/log has an entry of:

> SSDP create socket failed -6

### Actual results

No LAN devices are shown.

### Expected results

All LAN devices should be shown.

### Project file & Debug log uploads

[debug_network_Sat_Dec_02_19_52_45.log](https://github.com/bambulab/BambuStudio/files/13538019/debug_network_Sat_Dec_02_19_52_45.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

npike, 2023-12-03:
I have also tried:

1. Rebooting the Mac
2. Connecting both WiFi and ethernet on the mac
3. The latest release, 1.8.0 Public beta 2

bam-winterfell, 2023-12-04:
Hi @npike 
Sorry, we currently do not support multiple instances. We will fix this issue in the future

Satu7nine, 2025-01-22:
Given the upcoming changes and probable increases in the number of users using LAN or Developer mode, can this bug be reprioritised?

themoch, 2025-04-25:
I second this from @Satu7nine . As of `2.0.3.54` this is still an issue.

arekm, 2025-09-14:
@bam-winterfell this bug is still there in 2.2.1.60 (macos sequoia 15.6.1).

How to reproduce (in my case):
- start BambuStudio -> my A1 Mini (online), P1S (online) and X1C (LAN mode) show up correctly in Device tab.
- select P1S in device tab, BambuStudio connects to P1S correctly, everything is fine
- click "File"->"New Window"
- new window with BambuStudio shows up BUT **X1C (LAN mode) is not on the list in Device tab while it should be there**. Only A1 Mini and P1S are visible.
- close initial / first window
- you are **left with second window and X1C never shows up**
- the only fix is to restart BambuStudio to get X1C visible at least in first window, very annoying

Expected result: X1C (LAN mode) should always be visible in every instance.

Also please remove "inactive" label and add some "confirmed" instead.