bambulab/BambuStudio #3019: Crashes

Sundance3D, 2023-12-03:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

64-bit operating system, x64-based processor, Windows, 11 Pro build22621.2715

### Additional system information

Intel(R) Core(TM) i7-6700K CPU @ 4.00GHz   4.00 GHz
16.0 GB memory
64-bit operating system, x64-based processor
Windows 11 Pro
22621.2715

### Printer

Bambu Labs A1 mini

### How to reproduce

Often when I load a stl (mostly from sources other than Maker World but sometimes from there as well) the program crashes, just closes with not warning or error message.

### Actual results

Restart some times it's OK sometimes not.

### Expected results

the program to run without crashing.

### Project file & Debug log uploads

Don't know whereto get this

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2023-12-04:
@Sundance3D 

could you upload the log following this guide https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

also is it a 100% reproducible issue for all stls?
and did previous version work before?

thanks

Sundance3D, 2023-12-05:
Send how????????

No it occurs randomly with deferent stl's.
No previous version of Bambu studio installed.

bambu123, 2023-12-05:
> Send how????????
> 
> No it occurs randomly with deferent stl's. No previous version of Bambu studio installed.

How to upload studio log.
1. Open user data folder from the menu 'Help > Open Configuration Folder' in Studio
2. Zip the 'log' sub folder
3. Upload to github or somewhere else and provider the download link

If the zip file is too large, it is recommended to compress and upload the log separately. Please be sure to include the DMP files.

Sundance3D, 2023-12-06:
WHERE IS THE PRINTER LOG FILES??????? I am having trouble with my AMS and tried to send a ticket but you REQUIRE this file but GIVE NO CLUE WHERE IT IS LOCATED...what the heck????

bambu123, 2023-12-06:
> WHERE IS THE PRINTER LOG FILES??????? I am having trouble with my AMS and tried to send a ticket but you REQUIRE this file but GIVE NO CLUE WHERE IT IS LOCATED...what the heck????

Your question is about Studio Crash, so we need Studio's log files, including Crash DMP files, but not printer logs.

Sundance3D, 2023-12-06:
I have that information, it is for Bambu Studio. I am having issues with
the  AMS and requesting "printer logs". I don't see anything on the SD card
or the or in Bambu Studio called printer logs.

lanewei120, 2023-12-06:
for BambuStudio's log, please follow the guide provided by @bambu123 or follow this wiki https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

for printer log, 
please follow https://wiki.bambulab.com/en/x1/troubleshooting/export-failed-print-log-to-sd-card
and https://wiki.bambulab.com/en/x1/troubleshooting/upload-log to export log and create ticket

and for this issue, please help to upload the BambuStudio's log

thanks

Sundance3D, 2023-12-07:
This IS NOT for an A1 mini!

lanewei120, 2023-12-07:
> This IS NOT for an A1 mini!

this one for A1 mini
https://wiki.bambulab.com/en/a1-mini/troubleshooting/how-to-upload-log

but for this bug, as it is bambu studio related
could you upload the bambustudio's log here

Sundance3D, 2023-12-07:
YES, I am aware of that but I am trying to get a cross to you I am having issues with my AMS lite but in order submit a ticket I need the printer log file and I can't find it or know what to look for or where. I was told it is in the logger on the SD card but I see nothing about the printer and the files there do not work to allow me to make a new ticket about the AMS problem.

Thanks

SaltWei, 2024-01-18:
> Often when I load a stl (mostly from sources other than Maker World but sometimes from there as well) the program crashes, just closes with not warning or error message.

One ticket for one issue to avoid misunderstanding during communication.

So we return to this ticket.
@lanewei120 @bambu123 Have you got the studio log file of this crash issue from user? How about the solution

lanewei120, 2024-01-18:
not yet
@Sundance3D 
is your bambustudio still crash? if yes, please help to upload the log here
if not, we will close this issue

Sundance3D, 2024-01-18:
No it's OK now.

SaltWei, 2024-01-18:
> No it's OK now.

OK. Thank you very much for reply!
Please raise new ticket if you meet any other new issue.

Close this ticket now.

closed 2024-01-18 by SaltWei

Sundance3D, 2024-01-18:
I did see a bug in the latest version of Bambu Studio that just
downloaded tonight,
V1.8.4.51. *THIS IS NOT A TROUBLE TICKET......I AM JUST LETTING YOU KNOW IN
CASE YOU*
*ARE NOT AWARE OF IT.* See image below.
[image: redline.jpg]