bambulab/BambuStudio #3031: Problem with login

Bexxx78, 2023-12-04:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab A1 Mini

### How to reproduce

1. I open the Program
2. I would like to log in
3. The  Screen is white
4. The solution with the change of language, not working for me
[image]
[image]

### Actual results

The  Screen is white, i cant login

### Expected results

...

### Project file & Debug log uploads

[Dateien.zip](https://github.com/bambulab/BambuStudio/files/13550153/Dateien.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2023-12-05:
The problem is that the network plugin is not installed. Click on the Device page, follow the prompts to install the plugin, and then try logging in again.

Bexxx78, 2023-12-05:
Habe jetzt die

Bexxx78, 2023-12-05:
Actually I installed the plugin...strange.

I installed the other version of the program, this helped

Thanks

closed 2023-12-05 by Bexxx78