bambulab/BambuStudio #3034: X1 (and some P1) supports air filtration but are configured not to.

GuzziRaz, 2023-12-05:
### Bambu Studio Version

1.8.1.57

### Where is the application from?

Bambu Lab github releases

### OS version

macOS Sonoma 14.1.2

### Printer

X1C

### How to reproduce

Load a system preset for X1C

### Actual results

Printer is erroneously configured to not support air filtration and will not evacuate chamber.

### Expected results

The X1 series (and assumably P1 with chamber fan) does support air filtration. What is the point of supplying system presets specifying they do not?

This makes all system profiles unusable for anything but PLA. It forces user to edit them and always use user profiles. It's just annoying.

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

XunZhangBambu, 2023-12-06:
Sorry, this option will not affect regular printing. After enabling this option, you can choose whether to enhance filtration by adjusting the exhaust fan speed.

XunZhangBambu, 2023-12-06:
The original function of enhancing the filtration effect by controlling the exhaust fan speed was specific to the X1E and related params were tested based on X1E . Compared to other printers, X1E has a better filtration effect . So we only default enable the option for X1E . And for other printers , default disable the option will not affect the speed of chamber fan

GuzziRaz, 2023-12-06:
Fair enough. I guess on X1C we don't want the fan during printing because we'd lose temperature.

closed 2023-12-06 by GuzziRaz