bambulab/BambuStudio #3045: Beta Client keeps disconnecting from printer

Mathew-D, 2023-12-07:
### Bambu Studio Version

1.8.1.57

### Where is the application from?

Bambu Lab github releases

### OS version

Manjaro Sway Linux 6.6 kernel

### Additional system information

ryzen 2700
32GB Ram
Radeon RX 5600XT

### Printer

Bambu Labs X1C

### How to reproduce

Slice model.
Send Model to printer
Try and watch Device screen

### Actual results

The software sends the model to the cloud and opens the Devices tab but keeps connecting to the printer then failing to connect then connecting then failing again.  If you time it right you can get the video feed to work.  The video feed will keep going but the status on the printer will keep connecting then failing to connect but it does print.  I can watch the status of the printer on the Handy app just not on Bambu Studios

### Expected results

The status should stay on the screen.  If I use version 1.7.7 Bambu Studio it works fine.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/13593469/log.zip)
[SquirrelShield56deg.3mf.zip](https://github.com/bambulab/BambuStudio/files/13593478/SquirrelShield56deg.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Mathew-D, 2023-12-07:
Here is a link to a long screen capture for this.  https://streamable.com/aanqho

walterwongbbl, 2023-12-12:
Hi. @Mathew-D 
Thank you very much, I saw the video. Is the firmware version of the printer the latest

Mathew-D, 2023-12-12:
> Hi. @Mathew-D Thank you very much, I saw the video. Is the firmware version of the printer the latest

Yes the printer is running the latest version firmware.

norpol, 2023-12-16:
@Mathew-D Check if your computer has the right time set. I had the same issue and it turned out my network time protocol daemon didn't sync properly.

https://time.is/ should show max 1-2 seconds or best "your time is exact".

Mathew-D, 2023-12-16:
@norpol Thank you very much.. my time was off by 6mins.  Once I fixed the time daemon it seems to have fixed the problem!

closed 2023-12-16 by Mathew-D