bambulab/BambuStudio #3046: Bambu Studio keeps changing my selected printer without my consent

AdrianGarside, 2023-12-07:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1P & X1C

### How to reproduce

Not sure of the exact repro but it's something like this:

Open two instances of BambuStudio
Set one to the P1P and trigger a print.
Set second to the X1C and trigger a print.
Finish prints and go back to start next prints.

BUG:
The printer selected second randomly gets switched to in the first instance out of the blue. 

This behavior is driving me nuts.

### Actual results

Bambu Studio changes the printer on me out of the blue, randomly as far as I can tell (but I'm sure it's not random - just feels like it).

### Expected results

A running instance of Bambu Studio should *never* change the selected printer automatically.

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

AdrianGarside, 2023-12-07:
No these are both my printers and I am the only one with access to my account.

AdrianGarside, 2023-12-08:
So I think my initial interpretation of what was going on was incorrect. When I was seeing this I had all my instances of Bambu Studio arranged in the central 1/3 of my ultrawide monitor. To better understand things I repositioned the P1P instance in the left 1/3 and kept the X1C instance in the middle 1/3. 

And so far I haven't seen the printer change unexpected in those instances. However, I have seen other instances of Bambu Studio randomly jump to the foreground and steal focus. And since in my original repro I had the same 3mf loaded into both I think what actually happened is the 'other' instance randomly jumped to the foreground and stole focus making me believe that the printer had changed on me (since that was the only difference between the two instances). 

I have not yet worked out what is causing a background instance of Bambu Studio to jump to the foreground / top of the window stack.

walterwongbbl, 2023-12-08:
Hi @AdrianGarside 
We are unable to locate the issue based on description. If you encounter it again, please provide a screenshot or video. Thank you.

AdrianGarside, 2023-12-08:
App windows don't jump to the foreground and steal focus by themselves. You must have code there to do that somewhere. Maybe in response to a print completing?

walterwongbbl, 2023-12-08:
@AdrianGarside 
So you would like to switch to the window of the printer that has finished printing, is that correct?

AdrianGarside, 2023-12-08:
No, I don't want Bambu Studio to force a background instance of itself to the foreground and interrupt (and confuse) my interaction with the already foreground instance.

walterwongbbl, 2023-12-12:
@AdrianGarside 
Is everything okay so far?

AdrianGarside, 2023-12-15:
It's less obvious now that I keep my different instances separate as the instance stealing focus is no longer jumps on top of the one I'm interacting with. But I just split them across virtual desktops to make it even easier to manage the different prints I have going / want to queue next. And that made a consistent misbehavior obvious. When I start a print and move onto interacting with another instance, when the first instance switches to the device pane it steals focus. Which with virtual desktops in the mix is extremely obvious as that misbehavior now moves me to the virtual desktop that instance is on.

AdrianGarside, 2023-12-18:
Another observation. This also repros the problem:
1. Start a print in one Bambu Studio instance.
2. Switch focus to a non-Bambu Studio window and do some work.
3. Click on a different Bambu Studio instance later.
BUG: 
Instead of switching focus to the selected instance, it switches focus to the original Bambu Studio instance from #1. With virtual desktops in the mix this is *extremely* obvious.

AdrianGarside, 2023-12-18:
And it's even more amusing than that. I'd started two prints from two separate instances of bambu studio on the other virtual desktop. The second time I tried to switch focus to the 3rd instance on my main desktop _also_ stole focus and switched back to one of the instances on that second virtual desktop. Seems like there's somehow a queue of focus stealing being set up!

AdrianGarside, 2023-12-18:
Gah, no it's worse than that. I've now been switched to that 2nd virtual desktop 3 times! So it's not even 1:1 with print started.

walterwongbbl, 2023-12-19:
@AdrianGarside 
We will try to reproduce this issue

459below, 2024-01-06:
I have the same misbehavior on BambuStudio_linux_fedora_v01.08.02.56-20231213.AppImage

Any time the printer starts or finishes a print the window of Bambu Studio jumps to the top of the windows, ripping me out of what I am actually doing. And it does that two to three times each print.

walterwongbbl, 2024-01-18:
@459below 
Was the Bambu program minimized at that time?

459below, 2024-01-20:
The BambuStudio window just now jumped right to the front of my windows / focusmultiple times during a single print again.
It happens when the window is minimized, or not on top and even when it is on a separate virtual workspace. It happens when the print starts calibration and at least twice when the print has finishes.
I am using Gnome 3 here.

walterwongbbl, 2024-01-22:
@459below 
Thank you very much. We will try to reproduce this issue

glbyers, 2024-03-11:
I'm experiencing the same behaviour with Bambu Studio 1.8.4.51 installed from flatpak on Linuux & i have to say, it's really peeing me off. Have it running on my workstation while I'm trying to work on another virtual desktop, and every time there's some change, Bambu Studio steals focus. When doing a multi material print, this can happen 100s of times in the space of a few hours, making for an incredibly frustrating experience; like this computers sole task is to take care of Bambu Studio!

Please fix. It's driving me up the wall.

matt-fff, 2024-03-25:
I'm running into the same issue with the AMS. Constantly steals focus. I'll be typing in some other window and Bambu Studio interrupts what I'm doing - it usually takes me a bit to figure out why my inputs aren't working, since I have so many monitors.

The only current solution is to close Bambu Studio immediately after starting a print.

reefball, 2024-07-24:
I have exactly the same problem, I have 3 bambu printers and 3 monitors usually running multiple instances of Bambu Studio and the software behaves very badly not only stealing focus sometimes at seaming random times but it also leaves a lot of widows open from print notifications that have been resolved either on the printer screens themselves or by for example rebooting the printer, restarting a print etc.  It is quite annoying that it is difficult to close all these windows because unless the curser is over that particular closing X box for a very long time it won't regain it's focus on that particular window.  I think that somehow Bambu Studio is not running specific individual instances of the program when opened for a different printer and it thinks it's in the same program and get's confused with the focus.  Since I have 3 different monitors on my computer perhaps that is the common denomiator as I saw that others on this thread have similar setups.

gratux, 2024-08-06:
I am also experiencing this issue. I am using Arch Linux with the i3 window manager.

From what i can tell, when the application is on the "Device" tab, it will continuously steal focus and move on top of other windows on the same workspace. This means if i am typing a comment under this very Github issue with Bambu Studio open on my second monitor, my mouse and keyboard focus will jump to the slicer pretty much on every layer change (it seems to me it is related with the AMS actions on the right side of the window) and interrupt my typing.

Whether the camera stream is running or not does not change anything.
However, if i manually click out of the Device tab (for example i switch to Preview instead), this issue seems to stop.

Krakonos, 2024-11-06:
Same issue here, as others have reported. Ona actions (print finished, AMS changing filament, maybe something else), the app is foregrounded. If i'm in the middle of sentence, I write half of it to the studio before I realize what happened. Not to mention it's annoying as hell. But I'd still like to keep the studio open on a second monitor; now I need to close the studio completely and reopen when I want to check on it.

It might be a "feature", but the implementation is poor. I would suggest having tray notification that is less intrusive - but TBH it must not do any notification on AMS material change, since that would happen way too much and you don't need to know. Print finished or error is just about two things I want to know when I'm running a job. Running 1.9.7.52. Otherwise, I'm quite happy - everything works well so far. This and the fact I seem to be unable to use the studion in Lan only mode are the only two issues I have...

Sector14, 2024-11-25:
Same issue here running 1.9.7.52 at the moment on Linux. I have Bambu studio on a 2nd monitor to allow monitoring print via camera whilst doing other tasks on the first. Every time the filament changes or heaters kick in or nozel is wiped, bambu studio will steal focus.

There may be reasons you want this to happen, but it would be very useful to have a preference to totally disable stealing focus/activating the window/etc. Just in the time it's taken to type this issue comment, I've had to alt-tab back into the input box due to focus been stolen.

eprochasson, 2025-01-28:
Same issue in 2025, on Arch Linux, using version 1.10.1.50 and KDE.

How to reproduce: start a print with BambuStudio, switch to another window (any window, not just BambuStudio). Sooner or later, BambuStudio will steal the focus. This usually happens a few times.

mike-lloyd03, 2025-01-31:
Same here. It seems like it steals focus when the print starts, when starting bed leveling, when it starts the actual print, and when it finishes the print. This is incredibly disruptive to my workflow.

There's no reason for the app to ever steal focus without some user interaction (for example, Firefox takes focus when a user clicks a lick in another app.) If the app needs my attention for something, it should send a notification.

hadess, 2025-02-03:
Test patch is available at:
https://github.com/bambulab/BambuStudio/pull/5891

A Flatpak build is being run:
https://github.com/flathub/com.bambulab.BambuStudio/pull/48

Ko-fi page if you want to support my work on BambuStudio:
https://ko-fi.com/hadessuk

closed 2025-03-17 by lanewei120