bambulab/BambuStudio #3047: Missing "Brim-object gap" in latest 1.8 beta version?

charleyzhuyi, 2023-12-07:
### Bambu Studio Version

1.8.1.57

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

A1 Mini

### How to reproduce

1. Open Bambu Studio
2. go to "Others" Tab
3. Check out "Bed Adhesion" section

### Actual results

Missing "Brim object Gap" field in 1.8.1.57

[image]

### Expected results

Should have "Brim-object gap" per https://wiki.bambulab.com/en/software/bambu-studio/auto-brim

### Project file & Debug log uploads

[Falcon Heavy Launcher individual(3).zip](https://github.com/bambulab/BambuStudio/files/13594085/Falcon.Heavy.Launcher.individual.3.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2023-12-07:
the object gap option is visual on the develop mode
[image]
[image]
[image]

QingZhangBambu, 2023-12-07:
i check the version 1.8.1.57, the object gap option is visual
[image]
if you don't mind, could you upload log file to help us check what happened?

bambu123, 2023-12-07:
@charleyzhuyi You need to select "Advanced" mode to see this parameter.

closed 2023-12-07 by charleyzhuyi