bambulab/BambuStudio #3055: Poor print quality on non-overhangs when using slow-down-for-overhangs

AbeFM, 2023-12-07:
### Bambu Studio Version

1.7.7.8.9

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Printer

X1C-AMS (unimportant)

### How to reproduce

Slice a file.

### Actual results

Sliced with "bridge speed" at 50:
[image] 
Even at a 20 degree slope, this is using the bridge settings, not any of the overhang ones.
Bridge speed @ 0:
[image]
Still some slowing down, but not much.

Anyway, the issue is poor print quality on the full speed vertical sides of the part:
[image]
The speed, flow rate are identical (as best I can tell with poor coloring on chart), but the surfaces are really bad. I've seen this in many prints in the past, cause by 'slow down for overhangs".

This is in Bambu filament with default settings.

### Expected results

Sides should be unaffected by overhang code. There's a difference in layer-time, but outside of that I wouldn't expect a difference in finish.

### Project file & Debug log uploads

fdsdfsf
[BambuWhining.zip](https://github.com/bambulab/BambuStudio/files/13603600/BambuWhining.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Xiuqi-Huang-CN, 2023-12-08:
[image]
[image]
  Hello, AbeFM. We have checked you 3mf and find that the overhang speeds were set unreasonable: too slow.  In this kind of situation, the speeds of the defferent areas of the same layer vary so significantly, and can lead to flaws due to sudden changes of melt filament flowing. In addition, in most printing cases, when you set a much lower speed which will make the filament stay in the nozzle for a much longer time and get more heat, you are supposed set a lower nozzle temperature to make sure that the filament will not get overheated and flow unnormally. On the contrary, A higher printing speed requirs a higher nozzle temperature. Most importantly, the speeds and the nozzle temperatures needs to be set in a reasonable range: neither too low, nor too high.
    Besides, we printed your 3mf models and the same models with the default Generic PETG setting. The former looks bad just like what your picture shows: 
[image],
 while the latter looks very nice just as what we usually get:
[image]
[image].
   Hope that this can help you a little bit to learn more about and get more fun from 3D Printing.

Xiuqi-Huang-CN, 2023-12-08:
@AbeFM

AbeFM, 2023-12-09:
First off, thanks! That was a quick help with a print. I was already using some default profiles and getting results closer to yours.

Unless I'm missing, something, it looks like the defaults are quite slow:
[image]

but I had more luck with faster overhangs like you said:
[image]

And here's the slow 'default' settings (I had one extra 50 in 10-25% as shown)
[image]

[image]
Upper part: Very slow overhangs
Lower Part: Fast overhangs

Sorry about the 1,100,1,1 settings, those were trying to find what parts of the model would be effected in the slicer. **If there were a way to highlight the areas effected by each of those settings it would be a BIG usability improvement to fine tune overhangs.**

Can you share the exact profile you used?