bambulab/BambuStudio #3058: Manual Flow Rate Calibration: Inconsistent order of calibration prints

Schokobecher, 2023-12-07:
### Bambu Studio Version

1.80.62

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 14.0

### Additional system information

N/A

### Printer

Bambu Lab X1C

### How to reproduce

- Select _Calibration_
- Select _Flow Rate Calibration (Manual)_
- Print Pass 1 or 2

### Actual results

The printed objects follow a inconsistent order.
Especially Pass 2 makes no sense, while Pass 1 could be argued to be mirrored from the top value

### Expected results

The objects follow a consistent order.

**flowrate-test-pass1.3mf** and **flowrate-test-pass1.3mf** would need to be changed, the code extracts the flow rate based on its object name, so it shouldn't lead to complications.

### Project file & Debug log uploads

[image]
Pass 1 currently

[image]
Pass 2 currently

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Schokobecher, 2023-12-07:
[flowrate-test-pass2.3mf.zip](https://github.com/bambulab/BambuStudio/files/13606220/flowrate-test-pass2.3mf.zip)
I've changed the 3mf for Pass 2, but currently am mid-print. I'll test it once the printer is free again

zhimin-zeng-bambulab, 2023-12-18:
@Schokobecher Thanks very much! we will add this modification after testing.