bambulab/BambuStudio #3064: openexr corrupt patch

LaurentBerger, 2023-12-08:
### Bambu Studio Version

even with master

### Where is the application from?

Built from source by yourself or others

### OS version

windows 11 vs 2022

### Additional system information

13th Gen Intel(R) Core(TM) i9-13900KF 3.00 GHz NVidia RTX 3090

### Printer

waiting delivery

### How to reproduce

I tried to build dep

```
cmake .. -DDESTDIR=c:/lib/bambustudio_dep -DCMAKE_BUILD_TYPE=Release
cmake --build . --config=release

```

### Actual results

cmake  --build . --config=Release
MSBuild version 17.5.0+6f08c67f3 for .NET Framework

  Checking Build System
  Building Custom Rule C:/lib/BambuStudio/deps/CMakeLists.txt
  Building Custom Rule C:/lib/BambuStudio/deps/CMakeLists.txt
  Building Custom Rule C:/lib/BambuStudio/deps/CMakeLists.txt
  Building Custom Rule C:/lib/BambuStudio/deps/CMakeLists.txt
  Building Custom Rule C:/lib/BambuStudio/deps/CMakeLists.txt
  Building Custom Rule C:/lib/BambuStudio/deps/CMakeLists.txt
  Building Custom Rule C:/lib/BambuStudio/deps/CMakeLists.txt
  Building Custom Rule C:/lib/BambuStudio/deps/CMakeLists.txt
  Building Custom Rule C:/lib/BambuStudio/deps/CMakeLists.txt
  Building Custom Rule C:/lib/BambuStudio/deps/CMakeLists.txt
  Building Custom Rule C:/lib/BambuStudio/deps/CMakeLists.txt
  Building Custom Rule C:/lib/BambuStudio/deps/CMakeLists.txt
  Building Custom Rule C:/lib/BambuStudio/deps/CMakeLists.txt
  Building Custom Rule C:/lib/BambuStudio/deps/CMakeLists.txt
  Building Custom Rule C:/lib/BambuStudio/deps/CMakeLists.txt
  Building Custom Rule C:/lib/BambuStudio/deps/CMakeLists.txt
  Performing patch step for 'dep_OpenEXR'
CUSTOMBUILD : error : corrupt patch at line 5 [C:\lib\BambuStudio\deps\build\dep_OpenEXR.vcxproj]
C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(247,5): error MSB8066:
 la build personnalisée de 'C:\lib\BambuStudio\deps\build\CMakeFiles\fea2bd8c90cc09515bf127050616dcf7\dep_OpenEXR-patch.rule;C:\lib\
BambuStudio\deps\build\CMakeFiles\fea2bd8c90cc09515bf127050616dcf7\dep_OpenEXR-configure.rule;C:\lib\BambuStudio\deps\build\CMakeFil
es\fea2bd8c90cc09515bf127050616dcf7\dep_OpenEXR-build.rule;C:\lib\BambuStudio\deps\build\CMakeFiles\fea2bd8c90cc09515bf127050616dcf7
\dep_OpenEXR-build_debug.rule;C:\lib\BambuStudio\deps\build\CMakeFiles\fea2bd8c90cc09515bf127050616dcf7\dep_OpenEXR-install.rule;C:\
lib\BambuStudio\deps\build\CMakeFiles\81e85d90e05730ebabef268f74d76fa5\dep_OpenEXR-complete.rule;C:\lib\BambuStudio\deps\build\CMake
Files\b4ed71ba669bc48bb63b6224b78d8f2a\dep_OpenEXR.rule;C:\lib\BambuStudio\deps\CMakeLists.txt' s'est arrêtée. Code 128. [C:\lib\Bam
buStudio\deps\build\dep_OpenEXR.vcxproj]
  Building Custom Rule C:/lib/BambuStudio/deps/CMakeLists.txt
  Building Custom Rule C:/lib/BambuStudio/deps/CMakeLists.txt
  Building Custom Rule C:/lib/BambuStudio/deps/CMakeLists.txt
  Building Custom Rule C:/lib/BambuStudio/deps/CMakeLists.txt
  Performing update step for 'dep_wxWidgets'
  Building Custom Rule C:/lib/BambuStudio/deps/CMakeLists.txt
  Performing patch step for 'dep_wxWidgets'
  C:/lib/BambuStudio/deps/wxWidgets/0001-wxWidget-fix.patch:42: space before tab in indent.
[... 292 lines cut ...]

CUSTOMBUILD : error : patch failed: src/osx/cocoa/settings.mm:224 [C:\lib\BambuStudio\deps\build\dep_wxWidgets.vcxproj]
CUSTOMBUILD : error : src/osx/cocoa/settings.mm: patch does not apply [C:\lib\BambuStudio\deps\build\dep_wxWidgets.vcxproj]
C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(247,5): error MSB8066:
 la build personnalisée de 'C:\lib\BambuStudio\deps\build\CMakeFiles\8a4a75bcf26be3a22de1212b2da6d907\dep_wxWidgets-update.rule;C:\l
ib\BambuStudio\deps\build\CMakeFiles\8a4a75bcf26be3a22de1212b2da6d907\dep_wxWidgets-patch.rule;C:\lib\BambuStudio\deps\build\CMakeFi
les\8a4a75bcf26be3a22de1212b2da6d907\dep_wxWidgets-configure.rule;C:\lib\BambuStudio\deps\build\CMakeFiles\8a4a75bcf26be3a22de1212b2
da6d907\dep_wxWidgets-build.rule;C:\lib\BambuStudio\deps\build\CMakeFiles\8a4a75bcf26be3a22de1212b2da6d907\dep_wxWidgets-build_debug
.rule;C:\lib\BambuStudio\deps\build\CMakeFiles\8a4a75bcf26be3a22de1212b2da6d907\dep_wxWidgets-install.rule;C:\lib\BambuStudio\deps\b
uild\CMakeFiles\81e85d90e05730ebabef268f74d76fa5\dep_wxWidgets-complete.rule;C:\lib\BambuStudio\deps\build\CMakeFiles\b4ed71ba669bc4
8bb63b6224b78d8f2a\dep_wxWidgets.rule;C:\lib\BambuStudio\deps\CMakeLists.txt' s'est arrêtée. Code 1. [C:\lib\BambuStudio\deps\build\
dep_wxWidgets.vcxproj]

### Expected results

no error

### Project file & Debug log uploads

[BambuStudio-deps.zip](https://github.com/bambulab/BambuStudio/files/13619429/BambuStudio-deps.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-12-11:
When compiling again, you need to clear the build directory because patches for some modules cannot be applied twice.

LaurentBerger, 2023-12-11:
not sure that's the problem. As build process is very verbose it is difficult to get all message.
I delete everything and rebuild. It seems ok for dependency but for bambusudio it's not 👍 
```
C:\lib\BambuStudio\build>cmake .. -DBBL_RELEASE_TO_PUBLIC=1 -DCMAKE_PREFIX_PATH=C:\lib\bambustudio_dep\usr\local -DCMAKE_INSTALL_PREFIX=c:\install\bambustudio -DCMAKE_BUILD_TYPE=Release -DWIN10SDK_PATH="C:/Program Files (x86)/Windows Kits/10/Include/10.0.22621.0"
-- Building for: Visual Studio 17 2022
-- The C compiler identification is MSVC 19.35.32215.0
-- The CXX compiler identification is MSVC 19.35.32215.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.35.32215/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.35.32215/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: OFF
-- SLIC3R_DESKTOP_INTEGRATION: ON
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: ON
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: ON
-- CMAKE_PREFIX_PATH: C:\lib\bambustudio_dep\usr\local (from cache or command line)
-- PROJECT_SOURCE_DIR: C:/lib/BambuStudio
-- CMAKE_MODULE_PATH: C:/lib/BambuStudio/cmake/modules/
Building with Win10 Netfabb STL fixing service support
-- LIBDIR: C:/lib/BambuStudio/src
-- LIBDIR_BIN: C:/lib/BambuStudio/build/src
-- Boost toolset is unknown (compiler MSVC 19.35.32215.0)
-- Boost toolset is unknown (compiler MSVC 19.35.32215.0)
-- Boost toolset is unknown (compiler MSVC 19.35.32215.0)
-- Boost toolset is unknown (compiler MSVC 19.35.32215.0)
-- Boost toolset is unknown (compiler MSVC 19.35.32215.0)
-- Boost toolset is unknown (compiler MSVC 19.35.32215.0)
-- Boost toolset is unknown (compiler MSVC 19.35.32215.0)
-- Boost toolset is unknown (compiler MSVC 19.35.32215.0)
-- Boost toolset is unknown (compiler MSVC 19.35.32215.0)
-- Boost toolset is unknown (compiler MSVC 19.35.32215.0)
-- Found Boost: C:/lib/bambustudio_dep/usr/local/lib/cmake/Boost-1.78.0/BoostConfig.cmake (found suitable version "1.78.0", minimum required is "1.66.0") found components: system filesystem thread log locale regex chrono atomic date_time iostreams
-- Boost::boost exists
-- TBB found in C:/lib/bambustudio_dep/usr/local/lib/cmake/TBB
-- Found OpenSSL: C:/lib/bambustudio_dep/usr/local/lib/libcrypto.lib (found version "3.1.2")
-- Found CURL: C:/lib/bambustudio_dep/usr/local/lib/libcurl.lib (found version "7.75.0-DEV")
-- Found ZLIB: C:/lib/bambustudio_dep/usr/local/lib/zlib.lib (found version "1.2.13")
-- Found EXPAT: C:/lib/bambustudio_dep/usr/local/lib/expat.lib (found version "2.5.0")
-- Found PNG: optimized;C:/lib/bambustudio_dep/usr/local/lib/libpng16_static.lib;debug;C:/lib/bambustudio_dep/usr/local/lib/libpng16_staticd.lib (found version "1.6.35")
-- Found OpenGL: opengl32
-- FindGLEW: did not find GLEW CMake config file. Searching for libraries.
-- FindGLEW: GLEW_USE_STATIC_LIBS: ON.
-- FindGLEW: GLEW_INCLUDE_DIR: C:/lib/bambustudio_dep/usr/local/include
-- FindGLEW: GLEW_INCLUDE_DIRS: C:/lib/bambustudio_dep/usr/local/include
-- FindGLEW: CMAKE_FIND_LIBRARY_SUFFIXES for SHARED: .lib
-- FindGLEW: CMAKE_FIND_LIBRARY_SUFFIXES for STATIC: .lib;.a;.dll.a
-- FindGLEW: GLEW_SHARED_LIBRARY_RELEASE: C:/lib/bambustudio_dep/usr/local/lib/GLEW.lib
-- FindGLEW: GLEW_STATIC_LIBRARY_RELEASE: C:/lib/bambustudio_dep/usr/local/lib/GLEW.lib
-- FindGLEW: GLEW_SHARED_LIBRARY_DEBUG: C:/lib/bambustudio_dep/usr/local/lib/GLEWd.lib
-- FindGLEW: GLEW_STATIC_LIBRARY_DEBUG: C:/lib/bambustudio_dep/usr/local/lib/GLEWd.lib
-- FindGLEW: GLEW_SHARED_LIBRARY: optimized;C:/lib/bambustudio_dep/usr/local/lib/GLEW.lib;debug;C:/lib/bambustudio_dep/usr/local/lib/GLEWd.lib
-- FindGLEW: GLEW_STATIC_LIBRARY: optimized;C:/lib/bambustudio_dep/usr/local/lib/GLEW.lib;debug;C:/lib/bambustudio_dep/usr/local/lib/GLEWd.lib
-- FindGLEW: GLEW_LIBRARIES: optimized;C:/lib/bambustudio_dep/usr/local/lib/GLEW.lib;debug;C:/lib/bambustudio_dep/usr/local/lib/GLEWd.lib
[... 13 lines cut ...]
  vdb_print not found! OPENVDB_PRINT-NOTFOUND
Call Stack (most recent call first):
  cmake/modules/FindOpenVDB.cmake:317 (OPENVDB_ABI_VERSION_FROM_PRINT)
  CMakeLists.txt:549 (find_package)

CMake Warning at cmake/modules/FindOpenVDB.cmake:324 (message):
  Unable to determine OpenVDB ABI version from OpenVDB installation.  The
  library major version "" will be inferred.  If this is not correct, use
  add_definitions(-DOPENVDB_ABI_VERSION_NUMBER=N)
Call Stack (most recent call first):
  CMakeLists.txt:549 (find_package)

CMake Warning at cmake/modules/FindOpenVDB.cmake:345 (message):
  IlmBase::Half can not be found!
Call Stack (most recent call first):
  cmake/modules/FindOpenVDB.cmake:367 (just_fail)
  CMakeLists.txt:549 (find_package)

CMake Error at CMakeLists.txt:554 (message):
  OpenVDB could not be found with the bundled find module.  You can try to
  specify the find module location of your OpenVDB installation with the
  OPENVDB_FIND_MODULE_PATH cache variable.

-- Configuring incomplete, errors occurred!
```

bambu123, 2023-12-20:
@lanewei120 有没有遇到过？

lanewei120, 2023-12-20:
it seems openvdb is not build successfully when building deps

@LaurentBerger 
is there OpenVDB libraries and cmake files included in your C:/lib/bambustudio_dep？

for example
```
$ ls usr/local/lib/cmake/OpenVDB/
FindCppUnit.cmake  FindJemalloc.cmake  FindLog4cplus.cmake  FindOpenVDB.cmake  FindTBB.cmake  OpenVDBGLFW3Setup.cmake  OpenVDBHoudiniSetup.cmake  OpenVDBMayaSetup.cmake  OpenVDBUtils.cmake

$ ls usr/local/lib/libopenvdb*
libopenvdb.lib   libopenvdbd.lib
```

if yes, could you packed these files to us for a check

LaurentBerger, 2023-12-20:
YES
usr/local/lib/cmake OpenVDB.zip 
[OpenVDB.zip](https://github.com/bambulab/BambuStudio/files/13730073/OpenVDB.zip)
and YES 
lib
[libopenvdbd.zip](https://github.com/bambulab/BambuStudio/files/13730082/libopenvdbd.zip)

lanewei120, 2023-12-21:
@LaurentBerger 

hi, this issue is caused by deps building of OpenEXR

we have fixed it at https://github.com/bambulab/BambuStudio/pull/3188

please have a try

closed 2023-12-26 by lanewei120