bambulab/BambuStudio #3067: Slicing takes forever, CPU at 10%, no disc IO.

marcelopaniza, 2023-12-09:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Intel i7, gen 10, 32 gb ram.
video cards: AMD Radeon RX 6700 XT, Nvidia K2200

### Printer

Bambu Lab X1C

### How to reproduce

Slice the full cape of the attached.

### Actual results

It takes more than 1 hour to slice, CPU starts well at 100% than it is mostly at 10%.

### Expected results

It should slice within minutes.

### Project file & Debug log uploads

Attached the file above.

### Checklist of files to include

- [ ] Log file
- [X] Project file

Schokobecher, 2023-12-09:
You didn't attach any Project file or STL

marcelopaniza, 2023-12-09:
oh sorry... file is too big. Let me try reducing  size.

marcelopaniza, 2023-12-09:
Get the full cape fixed from here:
https://www.printables.com/model/391372-barbe-blanche-one-piece
Global support:
[image]
Object support:
[image]

Nozzle: 0.2:
[image]

Quality:
[image]

QingZhangBambu, 2023-12-11:
@marcelopaniza if you don't mind, could you upload a 3mf file, which could ensure we use the same parameter