bambulab/BambuStudio #3068: Prepare screen does not show correct AMS filament set when 2nd slot is empty but 1, 3 and 4 are full

BDJeff, 2023-12-10:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Labs X1C

### How to reproduce

Load slots 1, 3, and 4.  The Device screen shows the correct set of filament in the AMS, with slot 2 blank.  The Prepare screen does not.  It shows the filament rolls as being in slots 1, 2, and 3.  

### Actual results

The Device screen shows the correct set of filament in the AMS
 
[image]

The Prepare screen does not.
 
[image]

### Expected results

Prepare screen should show slot 2 as empty and what is shown as in 2 should be in 3 and what shown as in 3 should be in 4.

### Project file & Debug log uploads

No log needed, but since I am forced to include one, here is one.
[debug_network_Sat_Dec_09_17_26_14.log](https://github.com/bambulab/BambuStudio/files/13626618/debug_network_Sat_Dec_09_17_26_14.log)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-12-12:
@BDJeff 
In Filament, the indices of items and slots are not correlated, and they do not correspond one-to-one. Only non-empty slots are synchronized.

walterwongbbl, 2024-03-11:
This issue will be closed

closed 2024-03-11 by walterwongbbl