bambulab/BambuStudio #3072: Missing properties from sliced files

THE-SIMPLE-MARK, 2023-12-10:
**Is your feature request related to a problem? Please describe.**
I'm building a business using Bambu Printers, and I want to automatize the slicing process. However there are a few properties missing from the sliced gcode output which was available in the upstream repository of Bambu Studio and are required for me to properly calculate the prices of prints.

**Describe the solution you'd like**
I would like to get back the following proprties in the GCode output (header comments):
- filament used [mm] 
- filament used [cm3]

**Describe alternatives you've considered**
For me, I don't see any alternatives. I'm forced to use Prusa Slicer (the upstream repository) until this gets resolved, which means I need to convert config files, etc. etc.

WolfwithSword, 2023-12-11:
Definitely need this back in the gcode file generation. Currently using an old version of Bambustudio and orcaslicer to maintain functionality in third party stuff I've developed which relies on parsing the `filament used [g]` and `filament cost` lists.

Eldenroot, 2023-12-11:
Yes!