bambulab/BambuStudio #3073: Thumbnail image bug

KennyBeaudoin, 2023-12-10:
### Bambu Studio Version

1.8.1.57

### Where is the application from?

Bambu Lab github releases

### OS version

windows 11

### Additional system information

[image]

### Printer

x1c

### How to reproduce

the image of the thumbnail is not showing the buildplate

### Actual results

the image of the thumbnail is not showing the buildplate

### Expected results

the image of the thumbnail should show the build plate

### Project file & Debug log uploads

i dont have any files to upload

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-12-12:
@KennyBeaudoin 
This should be a feature.

walterwongbbl, 2024-01-18:
Hi. @KennyBeaudoin 
 the build plate is not displayed. Instead, we show a transparent background image with a grid. This is because some filament colors are too similar to the dark background, making it difficult to distinguish the model.