bambulab/BambuStudio #3075: D-Bus name does not follow naming guidelines

hadess, 2023-12-10:
### Bambu Studio Version

git

### Where is the application from?

Built from source by yourself or others

### OS version

Linux

### Printer

N/A

### How to reproduce

`com.bbl.*` is not a valid reverse DNS name with which to own a D-Bus service, as Bambu Lab doesn't own `bbl.com`.
It should be `com.bambulab.ProductName`. I would recommend using `com.bambulab.BambuStudio`.

### Actual results

BambuStudio owns a D-Bus name of `com.bbl.bambuslicer.InstanceCheck.ObjectXXX`

### Expected results

BambuStudio owns a D-Bus name of `com.bambulab.BambuStudio.InstanceCheck.ObjectXXX`

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [X] Log file
- [X] Project file

lanewei120, 2023-12-15:
already fixed in 1.8.2

closed 2023-12-15 by lanewei120

hadess, 2023-12-15:
> already fixed in 1.8.2

If you mention "Closes: #number` then github knows to tag the commit and close the issue when it's merged.

In this case, the bug was fixed in ece83b13c8b5524e43ec1974a060f7aa467afd5a. Thanks!