bambulab/BambuStudio #3076: Bambu Studio 1.8.0.62 Freezes when saving current setting

NVNDO, 2023-12-10:
### Bambu Studio Version

1.8.0.62

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

Bambulabs P1S

### How to reproduce

- Load any STL file
- Change the name of an already personalized and saved setting
- Press the "Save-button"
- Bambu Studio should freeze and eventually force you to close

Here are my settings of "0.20mm Standard Personalized" which I want to rename 
[image]

Trying to rename:
[image]

Frozen!!
[image]

It doesn't matter what Name - it freezes instantly.

### Actual results

Bambu Studio Freezes without saving - forcing to close application.

### Expected results

Saving the settings under its new name and keep on functioning. 

### Project file & Debug log uploads

[Z Seam issue.zip](https://github.com/bambulab/BambuStudio/files/13629260/Z.Seam.issue.zip)
[Safe_config_Freeze_Log_Data.zip](https://github.com/bambulab/BambuStudio/files/13629261/Safe_config_Freeze_Log_Data.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2023-12-11:
I tried it briefly, but it didn't reproduce. Please help me provide the DMP file when it freezes according to the following wiki.
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

NVNDO, 2023-12-11:
Hey @bambu123 unfortunately there're no dmp files created in the C:\Users\X\AppData\Roaming\BambuStudio\log but I created a zip-file from the DMP created by the task-manager, though its 181MB Big and I can't sent it over Github.

Sebazzz, 2023-12-16:
@bambu123 I also have a dump file. Where can I send the confidential link to it?

NVNDO, 2023-12-17:
I just updated to the recent version and until now I haven't had any issues... 

@Sebazzz what version are you running?

Sebazzz, 2023-12-17:
Same, running latest stable 1.8.x

Met vriendelijke groet,
Sebastiaan Dammann

bambu123, 2023-12-18:
> @bambu123 I also have a dump file. Where can I send the confidential link to it?

Sorry, we don't have official network storage. You need to use your own DropBox, OneDrive, etc.

Sebazzz, 2023-12-28:
> > @bambu123 I also have a dump file. Where can I send the confidential link to it?
> 
> Sorry, we don't have official network storage. You need to use your own DropBox, OneDrive, etc.

Yes, that's why I asked to whom I can send a confidential link with the memory dump. Since the memory dump contains everything the program had in memory, including likely API keys, I'm not going to put that link publicly on Github.

bambu123, 2023-12-28:
> Yes, that's why I asked to whom I can send a confidential link with the memory dump. Since the memory dump contains everything the program had in memory, including likely API keys, I'm not going to [...]

Please send to me: chunmao.guo@bambulab.com.

Sebazzz, 2023-12-28:
Thanks, I've just shared the link via OneDrive. Please let me know when you have it 👍

bambu123, 2023-12-28:
> Thanks, I've just shared the link via OneDrive. Please let me know when you have it 👍

Oh, no. My OneDrive account is diffirent, it's cm_guo@hotmail.com.

Sebazzz, 2023-12-28:
> > Thanks, I've just shared the link via OneDrive. Please let me know when you have it 👍
> 
> Oh, no. My OneDrive account is diffirent, it's [cm_guo@hotmail.com](mailto:cm_guo@hotmail.com).

That shouldn't matter, as you get an unique link regardless whether you have Microsoft account or not - but I reshared it anyway just now.