bambulab/BambuStudio #3078: Unable To Click Next When Dynamics Flow Calibration Gcode File Is Finished

sweetblackberry, 2023-12-11:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab P1s

### How to reproduce

Run a dynamics flow calibration and wait for the gcode to be printed, when the print task is done, UNABLE to Click Next

### Actual results

I got the correct K value from printing but can’t edit that because Bambu Studio only accepts the response from the internet or the printer, however, your Bambu studio has a bug.

### Expected results

1.Fix this bug.
2. Add a manual confirmation button like “ I confirm the dynamics flow calibration is done and I need to go to the next step”

the défaut setting of this function only wait for the feedback or response from the cloud/internet/printer is very buggy and annoying. 

Same issue reported here in #2364 

### Project file & Debug log uploads

No file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

NordicRavn, 2023-12-12:
same here. There is no button for next so i can save the result.

zhimin-zeng-bambulab, 2023-12-18:
@sweetblackberry  @NordicRavn  When you encounter this problem next time, could you provide the Studio log? Thanks!

NordicRavn, 2023-12-18:
> @sweetblackberry @NordicRavn When you encounter this problem next time, could you provide the Studio log? Thanks!

Of course! I have some new spools to test so i will come back.

sweetblackberry, 2023-12-21:
> @sweetblackberry  @NordicRavn  When you encounter this problem next time, could you provide the Studio log? Thanks!
https://github.com/bambulab/BambuStudio/assets/65060861/763f5ae3-8f98-46d1-bbf6-91311b802e46
When the flow calibration gcode file is done, the next button is grey, I'm unable to input the correct k value for third party filaments, this is a serious bug please fix it soon. The root cause is Bambu studio only relies on the response from the cloud, manual confirmation to the next step is missing.
请看视频，当流量校准的gcode 文件打印完成后，拓竹切片软件完全无法进入下一步，按钮都是灰色的，根本无法点击，软件完全依赖云端反馈但是实际尝试很多次都不行，请速速修改。
建议增加一个人工确认进入下一步的功能，因为你们的云端服务器经常有问题。
还有拓竹切片软件控制机器的界面，只能对软件自带的耗材(拓竹跟软件自带的少数第三方品牌)修改k值，却不能对用户自己添加的第三方耗材修改k值。

请速速修正这个bug.

sweetblackberry, 2023-12-21:
@zhimin-zeng-bambulab

polyphia, 2024-01-02:
I am having the same issue, tried 3 times in a row with latest version 1.8.26 and next button is always grayed out after the PA test print finishes .

mauricebish, 2024-01-04:
[image]
Same problem.

Version 1.8.2.56 on MacOS

zhimin-zeng-bambulab, 2024-01-16:
Could you have a try with firmware 01.05.01, it has improved this kind of problem.

zhimin-zeng-bambulab, 2024-02-01:
Does this problem still exist with firmware 01.05.01?

mauricebish, 2024-02-01:
No it has been fixedBest Regards,Maurice Bish

closed 2024-02-02 by zhimin-zeng-bambulab

eipekci, 2024-05-08:
same issue here 1.8.4.51 - 01.05.02.00 p1p