bambulab/BambuStudio #3082: Cannot create filament whose name begins with a number.

man55, 2023-12-11:
### Bambu Studio Version

1.8.0 Public beta 3

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab X1C Carbon Combo

### How to reproduce

Try to create a filament whose name begins with a number.

### Actual results

Cannot create filament whose name begins with a number.

### Expected results

We have a manufacturer called **3Dplast**. We would like to be able to create filament for it.

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

berezovskyi-oleksandr, 2023-12-11:
Same for me. Please fix, this is the manufacturer I use the most

berezovskyi-oleksandr, 2023-12-11:
Same for me. Please fix, this is the manufacturer I use the most

@DanBao-Bambu could you take a look?

DanBao-Bambu, 2023-12-12:
Yes, I know there is this restriction, and I'm sorry. I will check and change the rules, but please note that if all inputs are numbers, they will still not be allowed.

closed 2024-01-18 by DanBao-Bambu