bambulab/BambuStudio #3085: Windows Media Player is required fort this task! IN OTHER WORDS: WINDOWS USERS, INSTALL LEGACY CRAP TO MAKE STUDIO WORK.

Majestic7979, 2023-12-11:
### Bambu Studio Version

1.8.8.57

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

X1 Carbon

### How to reproduce

Use Windows 11, at least build 22621.2792 - try to play back webcam from Bambu Studio. See this PATHETIC message:

**
[image]
**

Bambu Lab, get a goddamn grip! This is not Windows XP/2002 anymore. In a similar, closed issue, you asked  a user to utilize DISM tool to add the feature. It is a LEGACY feature!

You've got a state of the art printer and you're asking people to install GARBAGE FROM THE EARLY 2000s? There is a reason Microsoft does not want to bundle this application as default and that is because it is DEAD.

Stop asking your users to use DEAD PRODUCTS to play back the webcam on modern versions of Windows.

Not gonna happen!

### Actual results

Can't play back webcam.

### Expected results

Play the video without stupid error messages expecting the user to enable a legacy, now unsupported feature from Windows XP/2001.

### Project file & Debug log uploads

Not required.

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2023-12-12:
This is a historical reason. The use of wxWidgets from the open source code of Prusa determines that the WMP backend is used for wxMediaCtrl on the Windows platform.
Refactoring the entire playback framework is still under planning, using ffmpeg + opengl may be a solution.

Majestic7979, 2023-12-12:
Please do because right now the webcam is not functioning on Windows and I won't (and nobody else should) install legacy components on such a beautiful, modern printer. It just makes Bambu Studio look cheap.

RastonWarrior, 2023-12-13:
I also had to re-install Windows Media Player to make this work, but OP should realize that old IE code is still deep in the Windows OS and OMG, have you seen the code for notepad?!  

Windows software very often requires windows crud.  You make a system as clean as you can, and when it comes to Windows that's all you can do.  Been that way since inception.  3D print software makers are standing on what work they can reuse, in this case Prusa. Are you going to refactor the playback framework yourself, then?

If not, I would prefer the dev's work on the incredibly laggy click-and-drag window latency bug, ticket no. #934 .  There are far more important tickets than this one.

Majestic7979, 2023-12-13:
> I also had to re-install Windows Media Player to make this work, but OP should realize that old IE code is still deep in the Windows OS and OMG, have you seen the code for notepad?!
>
> Windows [...]

Can you just STFU.

There are alternatives and I PAID FOR the printer, so obviously it's not my job to refactor anything, clown. Not gonna deal with people like you. I don't need some CLOWN coming to my ticket to tell everyone how much more important it is to them. This is VERY IMPORTANT TO ME.

Go mind your business in your own ticket.

dgponcegmail, 2024-03-03:
Same issue under a valid wondows10 pro.
Re installed Windows media player (old version) and new one from Windows Store.
Not able to see the webcam and I  located  the printer into a dedicated room away from my proximity. 

Now I manage to set up a virtualized windows11 in order to make it ran but as Majestic7979 said that is not the way to resolve it. Only a workaround until the issue gets fixed.

pomazanbohdan, 2024-06-29:
Windows 11 same error
Media component installed

[image]

probonopd, 2024-06-29:
After installation of WMP and a reboot(!) it started to work for me.

fahaaaxiki, 2024-12-06:
Hi， we have solved this issue in our latest Public version：[1.10.1 Public Release](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50), you can update to this version to check if it works

Majestic7979, 2024-12-06:
> Hi， we have solved this issue in our latest Public version：[1.10.1 Public Release](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50), you can update to this version to check if it works

Thank you, yes it is working, so I'm closing the issue and appreciate your company highly for listening to your customers. Final thing that would make BambuLab great would be to open the AMS RFID system to third party filaments. I paid full price for two units (over £600) and I am highly unhappy that you've locked the system. Please don't be like Hewlett Packard/Epson and treat this as some printer ink, it's not. It should work with normal NFC/MiFare tags so we can manually program our own specialist filaments to keep track of.

It really sucks that we have all the necessary hardware and software but you lock it down. BambuLab filaments don't cover all use cases and even so, they are not available in places like Brazil.

closed 2024-12-06 by Majestic7979

fahaaaxiki, 2024-12-10:
> Thank you, yes it is working, so I'm closing the issue and appreciate your company highly for listening to your customers. Final thing that would make BambuLab great would be to open the AMS RFID [...]

Thanks for your suggestion ,we will consider it seriously, and currently Studio has supported the function of Custom Filaments, you can use your third party filaments through this function.

Majestic7979, 2024-12-12:
> Thanks for your suggestion ,we will consider it seriously, and currently Studio has supported the function of Custom Filaments, you can use your third party filaments through this function.

No, Yeah, I get that there's custom filaments, but it makes no use of the AMS RFID system. That is what we all want for the hardware we paid full price for. If you're going to lock it down like some HP printer, then sell the AMS subsidized cost. I paid over £600 for two filament boxes with motors and few chips and you made it dumb with an arbitrary software lock. Please stop, you can still monetize filaments while opening up the system.

I would buy your filament if it was easy to get, priced like the competition, and delivered free and quickly, but none of those things apply. If I paid full price for AMS units you should then subsidize the filaments you sell! But you don't even have stock of them most of the time, and all we would need is some rewritable MiFare tags to make this work for our own other branded filaments. Please accept that your business model for the AMS is flawed and open up the RFID system. If you want people to buy your filament, then rather than alienating your customers you should focus on improving your business processes so you have high stock availability at all times, free shipping even for 1 roll, more colors and materials, and price it CHEAP, it's just plastic after all, not a gold nugget. I love and recommend your brand without reservations but you are not perfect yet, RFID being locked only to your own filament is stupid, sorry, but it is.

If you refunded me the £600 for the 2 AMS units I purchased then maybe I, and the majority of your customers, would not be so upset that it's locked. Again, if you want to imitate Hewlett Packard then give the AMS to users practically free and then make money from locked down filaments.

Please listen, we all want this to happen! 🥰

I am deeply unhappy that you locked it down with software for no reason than greedy business model. I want to be able to put some info on Bambu Studio for my custom filaments, slap a MiFare NFC tag on the spool, and you can write to the tag and keep track of the filament just like your own branded one.