bambulab/BambuStudio #3086: Bambu UI Scaling Error

coreycoldren, 2023-12-12:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Launch Bambu Studio
2. Connect to device
3. Initiate video of device
4. Resize Bambu Studio window

### Actual results

[image]

[image]

### Expected results

The Camera, Control, and Print in Progress panels should be resized appropriately and proportionately. Instead of the Camera panel sizing *way* too wide and messing with the scaling of the entire window.

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

LJKraus, 2023-12-14:
Sounds similar to https://github.com/bambulab/BambuStudio/issues/2574 which has been fixed in the 1.8 beta.

bam-winterfell, 2024-01-16:
hi@coreycoldren 
You can try using version 1.8.3 .
This issue should have been resolved already

LiZ-Li-BBL, 2024-01-16:
Get the latest version here https://github.com/bambulab/BambuStudio/releases/tag/v01.08.03.89

closed 2024-01-16 by LiZ-Li-BBL