bambulab/BambuStudio #3087: Custom filaments revert to ? in the AMS

mtwomey, 2023-12-12:
### Bambu Studio Version

1.8.1.57

### Where is the application from?

Bambu Lab github releases

### OS version

OXZ

### Printer

X1C

### How to reproduce

1. Create a custom filament
2. Select that filament in an AMS slot

### Actual results

Filament is selected, but about 5 seconds later, it reverts to unconfigured with a ?

### Expected results

Filament stays configured.

### Project file & Debug log uploads

No log

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2023-12-12:
Could you upload the log of the machine and Studio? We need that to target the issue.

marnovo, 2023-12-12:
Same here. Project doesn't seem to matter.
macOS 14.1.2, BS Beta 1.8.1.57, BL X1C 1.0.7.00.00, AMS 00.00.06.40

vndk36, 2023-12-12:
I do have the same issue here. Beta 3 of the slicer and last firmware on the machine. 

I cannot upload here the log as they are too big of a file now.

mtwomey, 2023-12-12:
My studio logs were also huge, so I clear them out and restarted studio and recreated the issue. It doesn't produce any logs - I think that is because I literally just open studio and select the filament and it happens. There is no project started and nothing done anywhere at all. I am getting the x1c logs for you though.

MrDB42, 2023-12-12:
Having same issue with the newest firmware and Bambu Studio Beta. 
[debug_Tue_Dec_12_13_25_36_17784.log.zip](https://github.com/bambulab/BambuStudio/files/13653053/debug_Tue_Dec_12_13_25_36_17784.log.zip)

Can't up load machine log file, to big. (376 MB)

DanBao-Bambu, 2023-12-13:
Okay, thank you for your log. I understand now. Will be fixed on the printer firmware.

QingZhangBambu, 2023-12-13:
will the custom filament be selected will displayed in AMS later? or just be "?" forever.

vndk36, 2023-12-13:
For my part, it stays with the correct custom name and colour for 10 seconds or so and then goes back to "?". I managed to rapidly click on the "prepare" tab of the slicer and I can see the correct custom filament in the filament list of the machine. However, after 10sec on this tab or the device tab, it reverts back to ?

MrDB42, 2023-12-13:
@DanBao-Bambu I know that it's two separate departments (software, firmware) but... can you ask the firmware department when a new firmware will be released to fix this issue?

man55, 2023-12-13:
Same here.
The only chance to select a custom filament is to do so directly from the printer screen.

MrDB42, 2023-12-13:
@man55 Weird. I couldn't even do that, until I did a firmware reset on the printer just now. Interesting...

man55, 2023-12-13:
Yesterday I managed to add two filaments. Moreover, one was successfully added from the Studio.
The second one was installed only through the printer screen, but the main thing is that the Studio was able to send it there.
But today I can't do anything. I create filament, but it doesn't even appear on the printer screen.

tokamac, 2023-12-14:
I have no AMS but I can tell your that Custom filaments (with the factor K in their PA profile) created in latest Bambu Studio (1.8.2.56) do not appear on the screen of my X1-Carbon (latest FW 01.07.00.00) even after power cycling. 

Moreover, when a Custom filament is selected in Bambu Studio in the _Device_ tab, going back to the _Preview_ tab to _Print plate_ from there, Bambu Studio automatically switches again to the _Device_ tab where the custom filament is no longer selected, only the "?" appears with no color, and the custom filament cannot be selected again manually as the print is currently processing.

Before clicking "Print plate":
[image]

After clicking "Print plate":
[image]

BTW? weir enough, the K factor still remains for the default filament despite no manufacturer and no plastic type have been selected???

Panda631, 2023-12-15:
I can confirm this bug.

thesainthell, 2023-12-15:
I have to confirm this bug too, @DanBao-Bambu any chance that the firmware department will create a hotfix for this an time soon? I mean, the custom filament feature introduced in 1.0.7.00.00 is simply not usable in this state.
I (but i guess we all) would appreciate any further Information on this.

hogum, 2023-12-15:
Same shit I got. T_T

OzDruiD, 2023-12-16:
Really strange, I cannot replicate this so happy to also provide my logs if required to help as a "Base line \ its working"

This is very strange, as I cannot replicate the fault, in full cloud only mode.
I have now created multiple new custom filaments., as primary and apepar on device tab and printer
Even creating a preset and then calibrating that preset (instead of the master) and then those also appear on the printer (k prorfile etc)

Studio version- 
[image]
Custom Filament setup
[image]

Studio - Device Tab - Test filament in Dropdown
[image]
Studio - Device Tab - Calibration available
[image]
and it stays there (Test TPU on external spool) - Device TAB\AMS section
[image]

KRiZ-R, 2023-12-16:
I also have this problem, but for now it seems it happens when adding an existing brand.
I have tried to add eSUN PETG and SUNLU PETG, both will not show on the printer.
When I added an non existing brand(non existing in Bambu Studio): REAL ABS, it showed up on the printer

DanBao-Bambu, 2023-12-16:
There seem to be two situations that can cause the Filament option on the printer screen to be displayed as "?". One is when connecting to a printer in the cloud, and the other is when connecting to a printer through a LAN. We are aware of the issue of "?" appearing on the printer screen when connecting to the printer in LAN mode. Another option is to see the custom Filament on the printer and select it, which will then change to "?" during cloud connected printing, right? @mtwomey @marnovo @tokamac @thesainthell @ Please confirm that if there is a "?" issue with the cloud connected printer, the printer's log is indeed required as I know it is relatively large

thesainthell, 2023-12-16:
@DanBao-Bambu Thanks for coming by. [Here is my Log-File](https://drive.google.com/file/d/1iWGV0_yUSDdzby8ACmCaf14-KEqwVViF/view?usp=drive_link). Hopefully this will help to figure the cause of this issue.

Here are some more information of my very last experience: Today I was able to assign an non existing vendor (Extrudr PLA) config to an AMS Slot (3) while an non Bambu profile (Eryone PETG CF) was also successfully loaded in an other AMS Slot (4). This situation worked until I changed the non Bambu Profile of Slot 4 to an other non existing vendor (Extrudr PETG) profile. Since then, I wasn't able to get it back working.

DanBao-Bambu, 2023-12-18:
After conducting my own testing and communicating with the firmware department, I have confirmed the following issues:
When Studio connects through the cloud, creating a new Filament and selecting it on the Device page of Studio does indeed result in a "?" after a period of time. The reason is that the printer screen has not received any update events or information about creating new materials, resulting in this situation.
The current solution is to create the Filament, go to the Materials tab on the screen, click to enter the slot setting interface of AMS, and the Filament information on the screen will be obtained from the cloud again. Afterwards, Studio can be used for AMS Filament selection.
I'm sorry for the inconvenience caused to you, and I will push for the repair of the problem.

DanBao-Bambu, 2023-12-18:
@KRiZ-R The problem you described seems different from others. When you create Filaments for eSUN and SUNLU, and enter the Filaments page of the printer screen to edit the Filaments, do you not have the Filaments you created under these two vendors? This may be because after creating a custom Filament in Studio, the cloud did not synchronize in a timely manner. You can exit the editing page and re-enter it. If it still doesn't work, make sure your "Auto sync user presets" option in Preferences is checked. If so, it may be a new bug. Can you send me your printer log and Studio log for analysis? thank you.

thesainthell, 2023-12-18:
> The current solution is to create the Filament, go to the Materials tab on the screen, click to enter the slot setting interface of AMS, and the Filament information on the screen will be obtained from the cloud again. Afterwards, Studio can be used for AMS Filament selection.

Can you make this more like a "step by step" guide? I've tried to reproduce this but without success. However, I cannot rule out the possibility that I simply did not understand your instructions.

Thanks in advance

OzDruiD, 2023-12-18:
@thesainthell  I believe this is what he means but @DanBao-Bambu  can jump in.

1. Using STUDIO - First, create a custom filament, configure as you will etc.  Do a calibration if you want etc
# You may have already done this..great!
## As long as you have cloud sync on and your printer is logged in, the next steps should work.

3. PRINTER - access the filament screen, then then AMS section and access a AMS Slot and select the filament you created in step 1.     - Select the PA\K profile drop down if you did it in step #1 too.  At this point, i find it best to ensure ALL slots are correctly set for the next step, if you sync in studio for example it will pull back all the currently assgined AMS slots.

4. STUDIO - If you "sycn,resycn, or select the AMS in the DEVICE TAB of STUDIO -  it will show and RETAIN the filament, retain than returning a ?  after a period of time.

As I cannot replicate the ? fault, I cannot test above,  but I believe there is some mis-information with studio not defining the filaments in the Printer, by hard-selecting it in the PRINTER - its then retaining it in the STUDIO SOFTWARE.

OzDruiD, 2023-12-18:
@DanBao-Bambu - I understand now why I thought I couldnt replicate it initially.
For me, once I select it in the PRINTER (AMS) screen, it doesnt reset anymore in STUDIO.     I can repeatidly select and de-select etc, close and reopen studio and studio seems to remember it once I select it in the printer "once".  Note:  I do have to select the K profile each time after closing\opening studio - it remembers the filament not the profile.

However, Yes If I create a new filament, then the same process occurs "?" until I at least select it once and confirm through the PRINTER screen.

tokamac, 2023-12-18:
This still doesn't work. I have jut created a custom PRUSAMENT filament in Bambu Studio then IMMEDIATELY went to the X1C (checking that the Wi-Fi connection is correctly established between the two in the Device tab): the Prusament filament never shows up in the manufacturer drop down menu, no matter I do on the printer screen or the tabs on the computer. It's getting a bit tiresome.

ppamment, 2023-12-18:
Same issue here. white ? when I assigned the custom filaments to the AMS slots. Using cloud mode.
Trying to assign anything from the printer screen isnt working, the custom filaments don't show up.

To make matters worse, I closed and reopened Bambu studio, and after carefully mapping all my custom filaments, they are no longer showing up in the "Custom filaments section", though they do all show up still in the filament selection dropdown, with their neatly created names made by the custom filament process??
I now can't even see them as an option to assign to an AMS slot on the Device page, nor are they selectable to try to re-add as custom filaments:

**"User presets" are all here, with the new names assigned when I created the custom filaments:**
[image]

**"Custom filaments" all gone after restarting Bambu Studio**
[image]

**Despite the user presets still being in the filament dropdown menu on the left, they can no longer be used to recreate the custom filaments for some reason**
[image]

**Also not available to add to the AMS any more:**
[image]

thesainthell, 2023-12-18:
@OzDruiD That helped me a lot to understand the problem better. 

I figured out that I have some sort of caching issue with an older, faulty filament preset that is preventing me from having success with the approach you described. During my test sessions, I created and deleted many filament presets (with the same information "Extrudr" "PLA" "NX2 Matt") until they suddenly disappeared. As of now, I can create a preset with the above information " vendor", "material" and "serial", but the preset no longer appears in the Studio Presets overview, nor in the AMS. After I create a new preset with a different vendor name "Extruder" but same other information, it works.

I can now create the preset and select it in the AMS as you mentioned. So I guess I was somehow successful.

OzDruiD, 2023-12-18:
So Presets - and Custom Filaments are different things.  Can make things confusing describing it as those, but I do assume you mean custom filaments created here 
[image]

OzDruiD, 2023-12-18:
> As of now, I can create a custom filament with the above information " vendor", "material" and "serial", but the filament no longer appears in the Studio "Custom Filament" overview, nor in the AMS. Seems to be the same problem @ppamment is having.
> 
> After I create a new custom filament with a different vendor name "Extruder" but same other information, it works.
> 
> I can now create the custom filament and select it in the AMS as you mentioned. So

So yeah  - seems a confirmed 2nd  \ slightly different issue -  imperfect "Syncing" and "syncing that reports back a ?"

KRiZ-R, 2023-12-18:
> @KRiZ-R The problem you described seems different from others. When you create Filaments for eSUN and SUNLU, and enter the Filaments page of the printer screen to edit the Filaments, do you not have [...]

What would you like me to test? And how can I send the logs.
I have noticed, that often when I create a Custom Filament, when I restart Bambu Studio, it has removed the new Custom Filament. The User preset is still there.

Other times, when I have created a Custom Filament, and restart Bambu Studio, it is still there, but the Custom Filament is not synced to the printer.

ppamment, 2023-12-18:
@OzDruiD Thanks for helping. Yes I got the difference between presets and custom filaments. I went through and converted all my presets into custom filaments using the "Copy current filament preset" option.
The custom filaments then all disappeared after restarting the slicer, and now, despite the presets still appearing in the preset section, I am no longer able to create custom filaments from them with the "Copy current filament preset" function, as they do not appear in the dropdown any more. I do not know why.
I took a look in my AppData folder and there are a load of JSON files that correspond to all the Custom filaments I created in the filaments folder:

[image]

They do not appear in the slicer though

OzDruiD, 2023-12-19:
> I went through and converted all my presets into custom filaments using the "Copy current filament preset" option.

Might pay to test a "Brand new" filament, using the Copy - but choose "Generic PLA" or soemthing else.
Also can open those JSON files and look for parent filamentr ID and see if that parent Filament still exists or you deleted it etc.,

Finally - Bambu will prolly want your files\logs.. so hold onto them for now.

DanBao-Bambu, 2023-12-19:
@ppamment Check where the preset JSON file you created is. I suspect that the JSON file you created does not currently have the correct "filament_id". If possible, could you share your log and user folders with me?

DanBao-Bambu, 2023-12-19:
I really hope to help you, but if it's just a problem, then I can only rely on my experience to determine what problem may have occurred, which is not accurate. Therefore, I hope that if you discover a problem, you can share the log folder at that time. Because a single log file may not detect the issue, it may already exist when creating custom materials, so the entire folder is required.
[image]

ppamment, 2023-12-19:
> @ppamment Check where the preset JSON file you created is. I suspect that the JSON file you created does not currently have the correct "filament_id". If possible, could you share your log and user folders with me?

I did not create any JSON files. If I did, then Bambu Studio created the, for me. I simply was anxious that my presets might have been lost and so checked my AppData and took a backup just in case. I have not manually changed any files in there.

Here are the logs and my user folder.

[log.zip](https://github.com/bambulab/BambuStudio/files/13715096/log.zip)
[3848803597.zip](https://github.com/bambulab/BambuStudio/files/13715098/3848803597.zip)

Thanks for the help. It's appreciated - I am in the middle of printing Christmas gifts etc and at a bit of standstill now.
I have to say that as a developer myself who has been responsible for many bugs in the past, I'm generally pretty understanding about these things, but it feels a little crazy to me that this bug was reported by users of the beta versions before the official release was put out, and yet the release went ahead anyway without addressing it...

ppamment, 2023-12-19:
@DanBao-Bambu
I was able to create steps to reproduce this issue:

1. Create new preset based on a system preset eg "Generic PLA". I savad it as "Generic PLA - Copy" the default
2. Create a "Custom filament", and select "Create from current preset" and choose "Generic PLA - Copy" as the preset to create it from. A new entry is created in the available filaments "{manufacturer} {filament_type} {serial} @{printer name} {nozzle_size} nozzle.
3. The filament "presets" list where you choose the filaments for AMS slots on the "Prepare" tab" now has a duplicate - "Generic PLA - Copy" and say "eSun PLA Basic @Bambu Lab X1 Carbon 0.4 nozzle". You don't need duplicates, so delete "Generic PLA - Copy".
4. At this point if you go to the device screen and try to assign eSun PLA Basic @Bambu Lab X1 Carbon 0,3 nozzle" to an AMS slot, you the white ? I understand from your posts that this can be worked around by visiting the filaments section on the physical printer screen before trying to assign the filement (is this right?). This appears to be issue A
5.  Close Bambu Studio. The  "eSun PLA Basic @Bambu Lab X1 Carbon 0.4 nozzle.json" still has a valid filament_id
6. Reopen Bambu Studio - "eSun PLA Basic @Bambu Lab X1 Carbon 0.4 nozzle.json" immediately gets the filament_id set to null and it is no longer found. This appears to be issue B

I walked into both issue A and issue B

Tried this twice with the same result both times

OzDruiD, 2023-12-19:
> * Create new preset based on a system preset eg "Generic PLA". I savad it as "Generic PLA - Copy" the default
> * Create a "Custom filament", and select "Copy current filament preset" and choose "Generic PLA - Copy" as the preset to create it from. A new entry is created in the available filaments "{manufacturer} {filament_type} {serial} @{printer name} {nozzle_size} nozzle.
> * The filament "presets" list where you choose the filaments for AMS slots on the "Prepare" tab" now has a duplicate - "Generic PLA - Copy" and say "eSun PLA Basic @bambu Lab X1 Carbon 0.4 nozzle". You don't need duplicates, so delete "Generic PLA - Copy".

@ppamment  i think this is the issue.
In creating a custom filament, I dont do the intial preset copy bit - there is no need .. and dont get duplicates to delete.
If your deleteing a filament preset that your creating the followup custom filament too, its possible then it will no longer be available after closing and opening Bambu studio - as it has become orphaned in the back end (If I understand what your saying correctly).

In the create screen - is this where your definting eSun as per your example?
[image]
resulting in
[image]

If you edit that cusstom filament - before deleteing anything like a duplicate, what does it show - this is what mine does.
[image]

ppamment, 2023-12-19:
@OzDruiD 

Yes, those are precisely the steps I took, and yes, the edit custom filament screen looked just like that when I was done. I even used the "Add preset" button on a couple of them as I had some existing presets for different diameters that I'd missed and didn;t add to the custom filament when creating it. That then led to having 2 entries, one for each nozzle diameter.

The first step I described - creating a preset based on a system filament is not something I would do going foraward to create new custom filaments. I had all my existing presets that I created pre v1.8 and I wanted to migrate over to become custom filaments in this case though. Copying all the settings by hand would have been tedious, which I assume is where there's an option to copy from a preset. I only added it into the reproducible steps to ensure it was a complete workflow from "stock" Bambu studio. You need to have a user preset for that flow to work as you cannot delete the system presets.

It's very possible the workaround for issue A would work for me, but I cannot try it now as the Custom filaments are no longer present in the slicer.

I do not know why issue B happens for me and not for youm

adrikm, 2023-12-20:
Confirming I have this same issue exactly as described

DanBao-Bambu, 2023-12-20:
@ppamment Thank you for your steps and logs. We should have found an issue that prevented the custom material from being found on the printer. The reason for this issue is that when you create a custom Filament based on a user preset derived from a system preset (such as "Generic PLA Copy"), the created preset will be considered to have a basic preset ("Generic PLA"), and this issue will be fixed in the next version.
The current solution is to create directly based on Filament or system presets, and then make parameter changes to the created presets. I know this can be troublesome, but I'm sorry, but it is indeed a solution.
Thank you again for your steps and logs.

DanBao-Bambu, 2023-12-20:
@OzDruiD I have reproduced this question, but I'm not sure if it's like @ppamment .
Whether to delete "Generic PLA Copy" is irrelevant, creating based on "Generic PLA Copy" is necessary.
After restarting Studio, Filament disappeared: Closing Studio, I deleted the preset that was created in my directory (for example: "eSUN PLA PPAM test @ Bambu Lab X1 Carbon 0.4 noise"). Start Studio and have it download the deleted preset from the cloud again. At this point, filling_ The ID will become null and it will not be displayed on the wizard page.

ppamment, 2023-12-20:
@DanBao-Bambu Just to clarify, I have not edited, moved, changed or deleted the json files in the directory directly. All interactions occurred via Bambu Studio.
Is it not at all possible to determine and populate the filament IDs? They look like hex values that presumably relate to something, but I do not know what. 
Failing that, could I remove the filament_id: null and add "inherit" to an appropriate system filament or something like that?

DanBao-Bambu, 2023-12-20:
@ppamment The function is available, but if the selected preset is a user preset created based on the system preset ("Generic PLA Copy"), it will not be available.
Additionally, deleting the field "filament_id" does not work.

OzDruiD, 2023-12-20:
> I had all my existing presets that I created pre v1.8 and I wanted to migrate over to become custom filaments in this case though. Copying all the settings by hand would have been tedious

Understood - makes perfect semse, i think its a bug fixed in next version (usiong pre-existing presets etc to make customs)

@DanBao-Bambu Sorry I am also confused - are you asking if I have this problem?   I dont. 
Maybe i waas confusing in replying to @ppamment with their example.
FYI- I also dont delete preset files "from the directory" - I handle everything in Bambu Studio itself, but I dont see this disappearing filament ID thing like @ppamment  does.

> I have reproduced this question, but I'm not sure if it's like @ppamment .
> Whether to delete "Generic PLA Copy" is irrelevant, creating based on "Generic PLA Copy" is necessary.
> After restarting [...]

DanBao-Bambu, 2023-12-20:
@ppamment After carefully analyzing the log you shared, I found that when you restart, data will be pulled from the cloud and the "fillment_id" in your file will be replaced with null. However, I did not reproduce this scenario. Can you still reproduce this problem? Or is this a must see for you?

thesainthell, 2023-12-20:
@DanBao-Bambu I do have this problem too. Unfortunately I can’t exactly say when this happened. But what I can bring into is the information that I am using Bambu Studio on two devices, one is on macos and the other is on windows. 
I do believe (sorry but I tested too much without any concept so I just can believe) this happend first on windows but because of the cloud sync it takes also effect on the macos machine. 
Maybe this brings a new perspective on this problem.

DanBao-Bambu, 2023-12-20:
@OzDruiD 
> Sorry I am also confused - are you asking if I have this problem? I dont.

I am explaining to you under what circumstances this issue may arise.

DanBao-Bambu, 2023-12-20:
Attention everyone: When selecting the "Copy Current Filament preset" option to create a custom Filament, if a user preset is selected instead of the root preset, it will cause problems. Therefore, this mode is not recommended for use temporarily.
Root preset: System preset and preset created using the "Create Filament" function. Excluding user presets saved as a save in the "Filament settings" interface. As follows:
[image]

OzDruiD, 2023-12-20:
> I am explaining to you under what circumstances this issue may arise.

Ahh , now reading it I think I understand, if the filament file **is deleted from directory** after \ before opening Studio - then I could expect to get that problem .  Thank you.

> Attention everyone: When selecting the "Copy Current Filament preset" option to create a custom Filament, if a user preset is selected instead of the root preset, it will cause problems.

Perfect- undertood.

tokamac, 2023-12-22:
Latest X1-C FW **01.07.01.00** does NOT correct the issue :(
(BTW where are the [release notes](https://wiki.bambulab.com/en/x1/manual/X1-X1C-AMS-firmware-release-history)? It's been hours the firmware has been out and they are not listed!)

east005, 2023-12-30:
I still have the same problem is there an estimation when it will be fixed ?

megothss, 2024-01-03:
I also have the same problem.

If I try to create a custom eSUN PETG Filament. The AMS reverts back to ? after 5 or so seconds and the custom filament is gone after restarting Bambu Studio.

The only workaround I found that worked was:

- Create a custom filament but do not use an existing vendor: I've called mine "eSun Custom".
  Everytime I tried to use the eSUN existing vendor, the filament wouldn't be synced to the printer console and it would be gone after restarting Bambu Studio.

- Once the filament is created. I selected it directly in the printer screen. 
  Trying to do this on the Device tav was reverting it to ?

ppamment, 2024-01-03:
@DanBao-Bambu Would be good to get an update. I'm back to Orcaslicer and have to recreate filament profiles from the backup i took as and when I want to print with each filament tyoe which is a real pain.

It's a major bug in the flagship feature of this BS release that was reported by beta users before the release, yet the public official release went ahead without fixing it, leaving myself and others in a big mess as I was trying to print christmas gifts etc. A little communication goes a long way to alleviate frustration at a time like this. I reallly hope its being scheduled into a bugfix release sooner rather thanlater and not being deferred until the next feaure release.

This feature was the one that convinced me to come back and try BS again after a year using Orca. I really look forward to it working as choosing the filament profile when I inset the spool into the AMS is definitely going to be one of the bigggest QOL improvements there has been. Shame too, as the motor noise cancellation which I has setup the previous day is IMO the other single most awesome addition since the printer was released, and I was so delighted (and amazed what a difference it made) to not have that constant motor noise in my life at high volume any longer.
🙏

DanBao-Bambu, 2024-01-06:
On version 1.8.2.56 and earlier versions of Studio, there was an issue with the workflow for creating Filaments. Below is a detailed introduction to the workflow:

Select "Copy Current Filament preset" as the creation method.
The selected preset is a user preset. As shown in the figure below.
[image]
This will result in the printer not being able to obtain the Filament, even if you enter through the Filament settings portal of the printer.

Use the first creation method "Create Base on Current Filament", and then select a system Filament instead of the already problematic Filament. After creating the Filament you want, make parameter changes to the newly created preset to achieve good printing results.

tokamac, 2024-01-06:
I have done many times what you are suggesting, i.e. use 1st creation method "Create based on current filament" creating various new filaments from basic default system filament presets like Generic PLA, Generic ABS, etc. then giving them a new name and the required parameters. These custom filaments NEVER appear on the screen of my X1-C in the dedicated pop-up menu. No matter what I do, power cycling, going back and forth in the menus, adding a K factor for the PA profile or not, etc. never a single custom filament has ever appeared on the screen of my X1-C, period.
(BYW no AMS here. I am using External Spool exclusively, which should act exactly in the same way)
The fact that Bambu Studio loses the custom filament with the ? marking when going back and forth from Device to Preview tab to Print plate is another bug, perhaps related.
This is quite puzzling that, considering your successive posts, the Bambu developers don't seem to understand what has been going on with this particularly nasty bug all this time.

DanBao-Bambu, 2024-01-06:
@tokamac Can you tell me your printer version number and Studio version number? Can you provide me with your Studio Log and User directory? I need to analyze them in order to locate your issue.

tokamac, 2024-01-06:
- Bambu Studio version: **01.08.02.56**
- X1-C firmware version number: **01.07.01.00**

I assume you are asking me to send you the folders "log" and "user" located on my computer at ~/Library/Application Support/BambuStudio/
If so, I can zip them but how can I send this small (8MB) file to you in a secure and private way?

DanBao-Bambu, 2024-01-06:
@tokamac My email is maosheng.wei@bambulab.com.

Naninani, 2024-01-06:
I dont think this is firmware related as it happens on p1p as well.
Imo this is a pure slicer + cloud bug.

[36 more replies]