bambulab/BambuStudio #3099: wxWEBVIEW_NAV_ERR_CONNECTION

dgauche, 2023-12-13:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma 14.2

### Additional system information

Apple MacBook Pro 16
Apple M1 Max 32GB

### Printer

BambuLab X1C

### How to reproduce

Open Bambu Studio

### Actual results

BambuStudio takes a **VERY LONG** time (many 10 minutes) to open.

When it finally opens, it provides the following error notice: 
```
An error occurred loading file:///Applications/BambuStudio.app/Contents/Resources/web/homepage/index.html?lang=en_US
wxWEBVIEW_NAV_ERR_CONNECTION
```

Here is the relevant snippet from the log file:
```
2023-12-13 06:49:58.092073[Thread 0x00000001dbf59000]:OnError: [wxWEBVIEW_NAV_ERR_CONNECTION] The request timed out.
2023-12-13 06:49:58.092195[Thread 0x00000001dbf59000]:06:49:58: Error; url='file:///Applications/BambuStudio.app/Contents/Resources/web/homepage/index.html?lang=en_US', error='wxWEBVIEW_NAV_ERR_CONNECTION (The request timed out.)'
```

[image]

### Expected results

The app should open and display the "homepage".
[debug_Wed_Dec_13_06_19_06_1654.log](https://github.com/bambulab/BambuStudio/files/13662008/debug_Wed_Dec_13_06_19_06_1654.log)

### Project file & Debug log uploads

[debug_Wed_Dec_13_06_19_06_1654.log](https://github.com/bambulab/BambuStudio/files/13662066/debug_Wed_Dec_13_06_19_06_1654.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2023-12-14:
Has this problem been recurring? Please refer to the Mac platform application not responding in the wiki below and provide relevant logs.
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

dgauche, 2023-12-30:
This problem is recurring. It happens EVERY time I start Bambu Studio.  Orca slicer on the other had has no issues and loads in less than 2 seconds.

I have also updated to Bambu Studio 1.8.2.56 and the same behaviour persists.

Once the loading has completed (around 10 minutes or so) it works fine.

dgauche, 2023-12-31:
I did a force quit when waiting for it to load.

Date/Time:        2023-12-31 10:00:21.991 +0200
End time:         2023-12-31 10:04:58.224 +0200
OS Version:       macOS 14.2.1 (Build 23C71)
Architecture:     arm64e
Report Version:   44
Incident Identifier: 6BE713E2-CEAD-43C1-8180-E39C7933199C

Data Source:      Stackshots
Shared Cache:     F9DDD844-7F3F-34BD-BE29-F0C72D5E5449 slid base address 0x187e28000, slide 0x7e28000 (System Primary)
Shared Cache:     AA32606F-3F8A-36DC-89AB-9C1BD7BF3104 slid base address 0x7ff80316c000, slide 0x316c000 (Rosetta)
Shared Cache:     2BF5B417-CD87-34D6-A876-2BBD0F25330F slid base address 0x1f5a44000, slide 0x75a44000 (DriverKit)

Command:          BambuStudio
Path:             /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:       com.bambulab.bambu-studio
Version:          01.08.02.56 ()
Team ID:          F4SKHPXDD9
Is First Party:   No
Architecture:     arm64
Parent:           launchd [1]
PID:              2679
Time Since Fork:  277s

Event:            hang
Duration:         276.23s
Duration Sampled: 1.70s (process was unresponsive for 275 seconds before sampling)
Steps:            17 (100ms sampling interval)

Hardware model:   MacBookPro18,2
Active cpus:      10
HW page size:     16384
VM page size:     16384

Time Since Boot:  546s
Time Awake Since Boot: 546s
Time Since Wake:  n/a (machine hasn't slept)

Fan speed:        0 rpm
Total CPU Time:   1.778s (3.0G cycles, 6.7G instructions, 0.45c/i)
Advisory levels:  Battery -> 3, User -> 2, ThermalPressure -> 0, Combined -> 2
Free disk space:  28.32 GB/926.35 GB, low space threshold 3072 MB
Vnodes Available: 76.57% (201496/263168)

Preferred User Language: en-ZA
Country Code:     ZA
Keyboards:        ABC
OS Cryptex File Extents: 7944

--------------------------------------------------
Timeline format: stacks are sorted chronologically
Use -i and -heavy to re-report with count sorting
--------------------------------------------------

Heaviest stack for the main thread of the target process:
  17  start + 2360 (dyld + 24800) [0x187ee50e0]
  17  main + 172 (BambuStudio + 395880) [0x1044e0a68]
  17  Slic3r::CLI::run(int, char**) + 29132 (BambuStudio + 70908) [0x1044914fc]
  17  Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 172 (BambuStudio + 12606356) [0x105085b94]
  17  wxEntry(int&, wchar_t**) + 128 (BambuStudio + 47040112) [0x10715c670]
  17  wxApp::CallOnInit() + 152 (BambuStudio + 46018720) [0x1070630a0]
  17  Slic3r::GUI::GUI_App::OnInit() + 16 (BambuStudio + 12695848) [0x10509b928]
  17  Slic3r::GUI::GUI_App::on_init_inner() + 5732 (BambuStudio + 12701656) [0x10509cfd8]
  17  Slic3r::GUI::MainFrame::MainFrame() + 2892 (BambuStudio + 12971880) [0x1050def68]
  17  Slic3r::GUI::MainFrame::init_menubar_as_editor() + 1968 (BambuStudio + 12981612) [0x1050e156c]
  17  Slic3r::GUI::MainFrame::FileHistory::LoadThumbnails() + 240 (BambuStudio + 13038540) [0x1050ef3cc]
  17  tbb::detail::r1::task_dispatcher::execute_and_wait(tbb::detail::d1::task*, tbb::detail::d1::wait_context&, tbb::detail::d1::task_group_context&) + 168 (BambuStudio + 42519980) [0x106d0cdac]
  17  tbb::detail::d1::task* tbb::detail::r1::task_dispatcher::local_wait_for_all<false, tbb::detail::r1::external_waiter>(tbb::detail::d1::task*, tbb::detail::r1::external_waiter&) + 1220 (BambuStudio + 42528776) [0x106d0f008]
  17  tbb::detail::d1::task* tbb::detail::r1::task_dispatcher::receive_or_steal_task<false, tbb::detail::r1::external_waiter>(tbb::detail::r1::thread_data&, tbb::detail::r1::execution_data_ext&, tbb::detail::r1::external_waiter&, long, bool, bool) + 676 (BambuStudio + 42534816) [0x106d107a0]
[... 250 lines cut ...]
                  17  tbb::detail::d1::start_for<tbb::detail::d1::blocked_range<unsigned long>, Slic3r::GUI::MainFrame::FileHistory::LoadThumbnails()::$_108, tbb::detail::d1::auto_partitioner const>::run_body(tbb::detail::d1::blocked_range<unsigned long>&) + 140 (BambuStudio + 13140316) [0x10510815c] 1-17
                    17  Slic3r::bbs_3mf_get_thumbnail(char const*) + 176 (BambuStudio + 2278868) [0x1046ac5d4] 1-17
                      17  Slic3r::_BBS_3MF_Importer::get_thumbnail(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>&) + 72 (BambuStudio + 2047280) [0x104673d30] 1-17
                        17  Slic3r::open_zip_reader(mz_zip_archive*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&) + 76 (BambuStudio + 7664032) [0x104bcf1a0] 1-17
                          17  mz_zip_reader_init_cfile + 144 (BambuStudio + 19540812) [0x105722b4c] 1-17
                            17  fseeko + 92 (libsystem_c.dylib + 80876) [0x18810ebec] 1-17
                              17  _fseeko + 732 (libsystem_c.dylib + 194220) [0x18812a6ac] 1-17
                                17  __srefill1 + 36 (libsystem_c.dylib + 47352) [0x1881068f8] 1-17
                                  17  _sread + 32 (libsystem_c.dylib + 47460) [0x188106964] 1-17
                                    17  __read_nocancel + 8 (libsystem_kernel.dylib + 14304) [0x1882277e0] 1-17
                                     *17  ??? (kernel.release.t6000 + 31592) [0xfffffe000853fb68] 1-17
                                       *17  ??? (kernel.release.t6000 + 1713832) [0xfffffe00086da6a8] 1-17
[... cut ...]
                                                           *17  ??? (kernel.release.t6000 + 5650340) [0xfffffe0008a9b7a4] 1-17
                                                             *17  lck_mtx_sleep + 228 (kernel.release.t6000 + 488072) [0xfffffe00085af288] 1-17
                                                               *17  ??? (kernel.release.t6000 + 579612) [0xfffffe00085c581c] 1-17
                                                                 *17  ??? (kernel.release.t6000 + 586204) [0xfffffe00085c71dc] 1-17

dgauche, 2024-01-01:
The issue has disappeared.

I explored creating custom filaments by following this article: https://wiki.bambulab.com/en/bambu-studio/create-filament but I did not actually create anything.

I check the General "preferences" (Bambu Studio --> settings) where I spotted that my region was set to EU while it should be set to "others".  I corrected this and things were a bit weird.

I closed and relaunched Bambu Studio and it opened in a few seconds. I needed to login again, but after that all was fine.

 I suspect that somewhere the region settings were changed by an update or similar.

closed 2024-01-01 by dgauche

bambu123, 2024-01-02:
From the log information, it seems that the loading of the 3mf thumbnail of the historical project is particularly slow, which may be related to the user area. We will continue to analyze it. Thank you.