bambulab/BambuStudio #3101: Rename filament settings

MrDB42, 2023-12-13:
**Is your feature request related to a problem? Please describe.**
It's frustrating when Bambu Slicer creates a new filament that it prepopulates the name. I would like to type something else that is more meaningful to me. Especially if I forget to add a descriptor or misspelled something.

**Describe the solution you'd like**
When editing the Filament settings page, I would just like to be able to edit the name field.

**Describe alternatives you've considered**
I've tried to go through the %username%/appdata/roaming/Bambustudio*/user/2397348106/filament folder and edit the appropriate files, but I just can't get it right.

**Additional context**
[image]

DanBao-Bambu, 2023-12-14:
The name of the main preset that has already been created cannot be changed, just like the system preset. Can you create some sub presets based on the current preset, which should meet your needs? For example:

https://github.com/bambulab/BambuStudio/assets/132529350/352a237e-1f6c-493e-afb0-cc41f9fde298

man55, 2023-12-14:
It would be great to be able to rename a filament preset.
I'm already a little tired of creating new presets and deleting old ones, just to change the name.

DanBao-Bambu, 2023-12-15:
Thank you very much for sharing. I will study them. At the beginning of the design, we did consider how to make the program simple and practical, but with the emergence of new features, we had to expose more concepts, such as "Filament" and presets, although we have tried to weaken their exposure as much as possible. The concept of Filaments is necessary because they are required to be used on AMS. Each preset needs to be under a certain Filament, and the name of the preset is used from the beginning to imply the design of the Filament.

Panda631, 2023-12-15:
I would second this demand!

ideaalab, 2024-02-15:
Yes, i have 10 child filaments from a parent one. Ive created this at the beginning when I didnt know the preset working. Now I would like to rename the base filament and its not possible. And I dont want to create one new parent plus 10 new child filaments!
Ive tried changing parameters inside files the files ( C:\Users\XXXX\AppData\Roaming\BambuStudio\user ), but they become overwritten as soon as I open Bambu Studio again.
An extra button for renaming would be simple to implement.

DanBao-Bambu, 2024-02-18:
This is actually not too simple, it involves cloud synchronization, AMS synchronization, the transformation of inheritance relationships in the Preset system, and so on.
If we want to implement this feature, I think we will not modify the Filament name in the Filament Settings, but in the Edit Filament interface.[image]
I will add this requirement to the requirement pool and develop it after evaluating its feasibility.

ideaalab, 2024-02-19:
I think child should link to the father by the father filament ID. As each filament has its own ID, no matter what the name is if the father change name, but the ID remains the same, the child will still point to the correct filament.

DanBao-Bambu, 2024-02-20:
"Filament_id" is not unique to every preset. For example, the "filament_id" of "Bambu PLA Basic @ X1C" and "Bambu PLA Basic @ P1P" are the same.

ideaalab, 2024-02-20:
Unless you implement an unique ID, then renaming properties of all the childs must be done also :(

Is there a manual way to do it? Ive tried renaming the parent and childs but gets overwritten as soon as I open BS

DanBao-Bambu, 2024-02-20:
The reason for being overwritten should be because it is synchronized with the cloud. You can try turning off cloud synchronization, but there are many functions that cannot be used.

DanBao-Bambu, 2024-02-26:
There is no solution to this, any changes made outside of synchronization will be synchronized to the cloud for consistency.

stiost, 2024-03-04:
Any progress on this? My one month old X1C is already becoming a mess of filament presets that I can't delete or rename because of "inheritance". At the very least let us configure the full name when creating new filament presets.

You also have some parts of the UI (e.g. under calibration) that enforces a 40 character limit but the create new filament dialog creates filaments with longer names.

I almost always end up with at least two filament presets per actual filament I add, that I can't delete or clean up. Arg.

DanBao-Bambu, 2024-03-04:
@stiost Hello, we plan to display the names of custom Filaments as aliases in future versions, just like the system preset. For example, if you create "XXX @ Bambu Lab X1C 0.4 noise", this preset will be displayed as "XXX" in the Filament selection box.

stiost, 2024-03-04:
That sounds like an excellent workaround, I can't wait! Thank you.

keithel, 2024-05-04:
> If we want to implement this feature, I think we will not modify the Filament name in the Filament Settings, but in the Edit Filament interface.

I think that's a fine way to implement it.
In terms of UI design within the Filament Settings dialog, here are some suggestions for how the UX should work:

1. Next to the tool buttons for Save and Delete, have another icon for "Rename". Clicking on this would turn the filament profile name combo box into an edit box (In Qt at least, that is all within the same control - Bambu Studio uses WxWidgets though - not sure what WxWidgets' Combo Box provides). Pressing <Enter> in the edit box would confirm the rename (but not commit it - that would occur when you press the save button I think).
2. Right click of the filament profile combo box, context menu with "Rename" menu item, behavior otherwise same as in (1).
3. Make the filament profile combo box more clearly a combo box like ou have for "Type" - where you have a combobox dropdown handle you click to open it. Then, if you click outside of the handle where the text is, it would trigger the combo box to display an edit field to edit the name, rather than opening the drop-down list.

Anyway - some ideas for how to implement the UX. I do realize some care in when things get committed needs to occur. Thus, I don't think you'd commit the change upon the user hitting <enter> after typing the name change, as the user may change their mind quickly, and having the cloud sync turnaround on that involved seems less than ideal.