bambulab/BambuStudio #3102: PETG sticks on nozzle during filament load

odospace, 2023-12-13:
### Bambu Studio Version

1.8.0 Public beta 3

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1 Carbon

### How to reproduce

- Start printing with PETG

### Actual results

The purged filament sticks to the nozzle or is thrown onto the plate.

### Expected results

It seems the the nozzle temperature during wipe and shake is too low for PETG (`nozzle_temperature_initial_layer[initial_no_support_extruder]-20`). 
Makes it sense to set the nozzle temperature during wipe and shake to `nozzle_temperature_range_high[initial_no_support_extruder]`?

### Project file & Debug log uploads

Two places within the machine start g-code using this "low" temperature setting
`M109 S{nozzle_temperature_initial_layer[initial_no_support_extruder]-20}`

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Xiuqi-Huang-CN, 2023-12-14:
Hello. Have you checked if the nozzle wiper works normally? If it is deformed orbad damaged, the nozzle wiping effect will be so bad that the filament can still stay on the nozzle even after wiping. And you can read this wiki to learn how to check it: [https://wiki.bambulab.com/en/knowledge-sharing/common-print-quality-problem]
[image]
Hope this can help you and wait for your further feedback.

Xiuqi-Huang-CN, 2023-12-14:
@odospace

odospace, 2023-12-14:
@Xiuqi-Huang-CN Thank you, i checked the nozzle wiper, it has a small pit:
[image]

closed 2023-12-14 by odospace