bambulab/BambuStudio #3109: Error on launch: add font of HarmonyOS_Sans_SC_Bold returns 1

joenotjim, 2023-12-14:
### Bambu Studio Version

1.8

### Where is the application from?

Bambu Lab github releases

### OS version

openSUSE Tumbleweed 20231204

### Additional system information

AMD Ryzen 7 PRO 3700U w/ Radeon Vega Mobile Gfx
40 GB DDR4

### Printer

Bambu Lab X1C

### How to reproduce

Launch AppImage

### Actual results

```
./Downloads/BambuStudio_linux_fedora_v01.08.02.56-20231213.AppImage
[2023-12-14 10:22:20.608904] [0x00007f5c3bd92d40] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1

(bambu-studio:24072): GLib-GObject-CRITICAL **: 10:22:20.858: invalid (NULL) pointer instance

(bambu-studio:24072): GLib-GObject-CRITICAL **: 10:22:20.858: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
```

### Expected results

Software should launch

### Project file & Debug log uploads

```
./Downloads/BambuStudio_linux_fedora_v01.08.02.56-20231213.AppImage
[2023-12-14 10:22:20.608904] [0x00007f5c3bd92d40] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1

(bambu-studio:24072): GLib-GObject-CRITICAL **: 10:22:20.858: invalid (NULL) pointer instance

(bambu-studio:24072): GLib-GObject-CRITICAL **: 10:22:20.858: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
```

### Checklist of files to include

- [ ] Log file
- [ ] Project file

joenotjim, 2023-12-14:
I downloaded the Fedora AppImage (since openSUSE is RPM-based) two weeks ago. After three rounds of chasing missing libraries, the software launched and appeared to operate successfully. I don't actually have my printer yet, so I didn't do anything beyond opening the software.

After installing patches, I now get the error I reported in the bug. I downloaded and installed the HarmonyOS_Sans_SC fonts but the error persists. Could be a red herring.

Launching the AppImage with `--debug 5` does not produce any additional log data.

beanotj, 2024-01-09:
Same issue - I have a machine that was not updated and it still works fine.  I also tried 1.7 and all versions are broke on Opensuse Tumbleweed after the system updates.

lanewei120, 2024-01-16:
@beanotj @joenotjim 

sorry for the late reply

how about these images；
https://flathub.org/apps/com.bambulab.BambuStudio
or
https://github.com/bambulab/BambuStudio/releases/download/v01.08.03.89/BambuStudio_linux_fedora_v01.08.03.89-20240109.AppImage
or
https://github.com/bambulab/BambuStudio/releases/download/v01.08.03.89/BambuStudio_linux_ubuntu_v01.08.03.89-20240109.AppImage

joenotjim, 2024-01-16:
Both AppImage files are still broken, same error. The Flatpak works though! I should also mention that this issue appears to be from upstream. I get the same error in OrcaSlicer and Prusa Slicer doesn't work either, but with a different error.

lanewei120, 2024-01-19:
@beanotj @joenotjim 
have you installed webkit2gtk4 or webkit2gtk3 ?

joenotjim, 2024-01-19:
webkit2gtk4_0-37

Based on my update history, it looks like I had webkit2gtk4_1 v2.42.1-4.1 when I first tried to run the AppImage. I got an error about a missing library that I did not record, but as a result, I installed libwebkit2gtk-4_0-37, which was at v2.42.2-2.1 and resolved the error.

A week or so later, I ran updates, where the following packages:
libwebkit2gtk-4_1-0
webkit2gtk-4_1-injected-bundles
libwebkit2gtk-4_0-37
webkit2gtk-4_0-injected-bundles
were all updated to v2.42.2-3.1, at which point I started getting the null pointer error. They have subsequently been updated to v2.42.4-1.1, and I don't have an easy way to restore the working 2.42.2-2.1 version.

Thank you.

lanewei120, 2024-08-08:
Hello, @joenotjim 
still has issues with those app images?

[image]

matusnovak, 2025-10-15:
I have the same issue with 2.2.2 Public Release on EndeavourOS (6.12.34-1-lts)