bambulab/BambuStudio #3110: verbindung  zum drucker im bambustudio

Bulli1989, 2023-12-14:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Additional system information

CPU: AMD ryzen 7 3700x
16GB RAM
Gpu: 3060ti

### Printer

Bambu Lab x1c

### How to reproduce

Der drucker verliert immer wieder erneut die verbindung im bambu studio

https://github.com/bambulab/BambuStudio/assets/153852229/ea9fd790-0d07-4e3c-ade1-809ee9815c87

### Actual results

Siehe Video

### Expected results

Stabile verbindung

### Project file & Debug log uploads

https://github.com/bambulab/BambuStudio/assets/153852229/bd80da29-07ee-4962-849a-6d97064d9bed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-12-15:
Hi @Bulli1989 
Please provide the log files for the studio and printer, and we will assist you in resolving the issue.

Bulli1989, 2023-12-15:
Und wie genau mache ich das für beides? Habe das noch nie gemacht!--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

c-gloeckl, 2023-12-30:
Is there already a solution for this problem?
I've had the same problem since yesterday.
G
CHRIS.

kekstuete78, 2023-12-30:
find in an other treat ...
Problem solved:
synchronize system time/clock

In einem anderen Forum gefunden ...
Problem gelöst:
synchronisiere die Systemzeit/-uhr des PC

c-gloeckl, 2023-12-31:
> find in an other treat ... Problem solved: synchronize system time/clock
> 
> In einem anderen Forum gefunden ... Problem gelöst: synchronisiere die Systemzeit/-uhr des PC

thanks so much
was really the right solution
I would never have thought of that

Vielen vielen Dank 
War echt die richtige Lösung
Da wäre ich nie drauf gekommen

closed 2024-02-04 by walterwongbbl